raspberrypi/pico-feedback #188: pico REPL on uart (chapter 2.2 pico-python-sdk)

villeneuve, 2021-08-23:
The pdf documentation "Raspberry Pi Pico Python SDK" chapter 2.2 page 9 says:
to expose REPL over UART change in mpconfigport.h 
**#define MICROPY_HW_ENABLE_UART_REPL             (0)**
to 
**#define MICROPY_HW_ENABLE_UART_REPL             (1)** 
and build a new firmware (chapter 1.3)

However I found it's not enough we must also disable REPL over USB.
In the same file change:
**#define MICROPY_HW_ENABLE_USBDEV                (1)**
to 
**#define MICROPY_HW_ENABLE_USBDEV                (0)** 

Failing to do it gave me unusable pico, no REPL over UART nor over USB (I was expecting to have REPL on both: uart and usb)
When disabling REPL over USB  I had the REPL over UART as expected.

Thank you

aallan, 2021-08-23:
> However I found it's not enough we must also disable REPL over USB.

That should not be the case. You can have REPL over both UART and USB at the same time AFAIK.

villeneuve, 2021-08-23:
You are right! I tried again and have REPL over both UART and USB at the same time .
I should have messed up somewhere. Sorry for the inconvenience.
And thank you very much for your work and the very nice Raspberry Pi eco-system you are giving us :-)