raspberrypi/pico-feedback #189: Software serial and PWM functions?

xiahualiu, 2021-09-06:
So I have not programmed the Pico yet, but I have a few questions about the SDK first, I can be wrong so you can correct me if I am wrong.

I saw the SDK has hardware UART and pwm APIs, but I did not find the software UART and PWM APIs.

Is there any way to do software serial and pwm communication over GPIOs? This should not be a problem on the hardware.

The reason is because I have 5 sensors, they all have serial interfaces. Unlike SPI and I2C, serial and pwm is 1-to-1 so I need 5 serial ports.

Thanks!

Wren6991, 2021-09-06:
I moved this to pico-feedback because it is not a bug with the SDK.

aallan, 2021-09-06:
This is the sort of question you should be asking on the [forums](https://www.raspberrypi.org/forums/), so I’m closing this issue. But please follow up with further comments if you have specific questions about the [documentation](https://www.raspberrypi.org/documentation/).

closed 2021-09-06 by aallan