raspberrypi/pico-feedback #190: DMA registers marked as RO in documentation, SDK writes them

mgottschlag, 2021-10-03:
The DMA register aliases are marked read-only in the documentation (and probably in the .svd as well, which is why the API of https://github.com/rp-rs/rp2040-pac does not allow modifications). The official SDK, however, writes them when required to generate triggers.

The documentation should be updated accordingly.

#162 is a very similar issue.

aallan, 2021-10-04:
Both #162 and #110 are linked I think? (both of these link to our internal issue SPIV-60). Another one for the pile @lurch ?

kilograham, 2021-10-04:
I just noticed this in the context of https://github.com/raspberrypi/pico-sdk/pull/583

lurch, 2021-10-05:
> Another one for the pile @lurch ?

This one is slightly different in that it's specific to (the aliases in) the DMA block, so I don't think SPIV-60 applies here?