raspberrypi/pico-feedback #191: Documentation: `IC_SLV_RESTART_DET_EN` is absent for the datasheet.

ithinuel, 2021-10-04:
`IC_SLV_RESTART_DET_EN` is mentioned on page 499 in

> Indicates whether a `RESTART` condition has occurred on the I2C interface when `DW_apb_i2c` is operating in Slave mode and the slave is being addressed. Enabled only when `IC_SLV_RESTART_DET_EN=1`.

but does not seem to appear anywhere else.

lurch, 2021-10-04:
See https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2040/hardware_structs/include/hardware/structs/i2c.h#L108 and the comment just above it which says

> // List of configuration constants for the Synopsys I2C hardware (you may see
// references to these in I2C register header; these are *fixed* values,
// set at hardware design time):

@Wren6991 / @kilograham Is it maybe worth including that list in the datasheet too, rather than only buried as a comment in one of the `pico-sdk` header files? :man_shrugging:

aallan, 2021-10-04:
At minimal we should insert the comment as a `[NOTE]` somewhere appropriate in the datasheet.