raspberrypi/pico-feedback #198: Pico will not mount as a drive in BOOTSEL mode

Tomwhiting1987, 2021-11-01:
I have managed to stop my Pico from showing up as a mounted drive in BOOT mode. This happened after I changed the drivers using Zadig to the CDC SErial ones on RP0 and FP1 Interfaces. Any ideas how I can change these back so that the Pico shows up as a drive again?

lurch, 2021-11-02:
What OS and version are you running?

Tomwhiting1987, 2021-11-02:
it's Windows 10

lurch, 2021-11-03:
I once saw similar (ish?) behaviour when I was testing a new feature in a beta version of picotool. I've just rummaged through my old notes and found:

> I eventually found the "fix" for this is to go into Device Manager, "Update Driver", "select a driver from my computer" and then select "USB composite device".

closed 2021-11-16 by aallan