raspberrypi/pico-feedback #202: Measuring latency between PIO SM and DMA

rhulme, 2021-11-09:
See:
https://github.com/rhulme/picoROM_pio/issues/2#issuecomment-962603752

I measure 11 cycles in a PIO state machine that has just auto-pushed an address to its RX FIFO and is stalled waiting for a byte to be placed in its TX FIFO.  The RX FIFO will be read by a primed DMA channel to write into a second channel's 'read' trigger, which in turn will copy a byte from RAM into the SM's TX FIFO (so two single transfers)

Assuming 1 cycle for the `out` instruction, where are the other 10 cycles being used?  Is there any latency between writing to one of the DMA trigger registers and the transfer actually taking place?

I understand figure 13 in the RP2040 datasheet to mean there is a 1 cycle delay after DREQ has been detected before the DMA transfer starts (so one cycle to assert DREQ, one cycle delay and then the transfer starts)

Is the DREQ for a SM's RX FIFO asserted in the same cycle as the ISR is pushed into the FIFO or in the cycle afterwards?

Wren6991, 2021-11-09:
[image]
DMA transfers go through four phases within the DMA (which overlap for four concurrent transfers, giving 1 word/cyc throughput):

- Read address issue
- Read data completion
- Write address issue
- Write data completion

This leads to four latency cycles between a transfer being triggered, and being "seen" at the write destination. In your scheme it sounds like you are incurring this latency twice, which is responsible for 8 of your cycles.

rhulme, 2021-11-10:
Thank you for the quick response.

Just to be sure I've understood - transferring one word (8/16/32 bits depending on configuration) will take four cycles (one for each of the phases you mentioned).  Any DMA transfer for a single word will take four cycles.  But transferring two words would take five cycles, not eight because once the first word moves to the "read data completion" phase, the second word starts the "read address issue"?

So a DMA transfer of N words will take 4 + (N - 1) cycles?  Once you reach the first "write data completion" phase, you're now copying 1 word/cycle?

lurch, 2025-07-09:
Not quite the same thing that's being asked about here, but you may also find https://github.com/raspberrypi/pico-feedback/issues/280 useful.