raspberrypi/pico-feedback #203: Picoprobe with openOCD on RPI-400

LummoxMakes, 2021-11-11:
Hello hello 

Getting Started with RPI pico pdf
build-date: 2021-11-04 
build-version: 150df05-clean
Page 64
[image]

Discussed here 
[https://forums.raspberrypi.com/viewtopic.php?p=1935038&hilit=libusb#p1935038](url)

I have downloaded the latest updates discussed here 
[https://forums.raspberrypi.com/viewtopic.php?p=1932859&hilit=sdk+update#p1932859](url)

Using 
Getting Started with RPI pico pdf
Chapter 1 Quick Start guide to rebuild the tools chain

```
pi@raspberrypi:~/pico/picoprobe/build $ openocd -f interface/picoprobe.cfg -f target/rp2040.cfg -s tcl
Open On-Chip Debugger 0.11.0-g4f2ae61-dirty (2021-11-11-08:17)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
adapter speed: 5000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: Failed to open or find the device
Error: Can't find a picoprobe device! Please check device connections and permissions.
```
However this works ok
`sudo openocd -f interface/picoprobe.cfg -f target/rp2040.cfg -s tcl
`

This is my first post, I am new to RPI-400 and the RPI OS so please be kind.
Thank you

lurch, 2021-11-11:
@liamfraser @kilograham Is this related to https://github.com/raspberrypi/openocd/pull/5 somehow?

kilograham, 2021-11-11:
it is entirely related... try following the instructions at the top of that file

Here from the OpenOCD README

> For USB devices on GNU/Linux you should use the contrib/60-openocd.rules
file. It probably belongs somewhere in /etc/udev/rules.d, but
consult your operating system documentation to be sure. Do not forget
to add yourself to the "plugdev" group.

https://github.com/raspberrypi/openocd/blob/rp2040/contrib/60-openocd.rules

I wrote something similar in the forum issue, however perhaps someone else can explain it better

LummoxMakes, 2021-11-11:
Hello Hello 

Thank you for your immediate reply, however as documented on the same forum post you mentioned this solution does not resolve the issue and the reason for the escalation to github

The application note is outstandingly good but for this one issue on the RPI OS.

LummoxMakes, 2021-11-12:
Hello hello 

Updated RPI OS (32bit) Buster with this mornings update however the issues remains 

I also updated the toolchain by using the following 

Getting started with RPI Pico 
Page 4
wget https://raw.githubusercontent.com/raspberrypi/pico-setup/master/pico_setup.sh
$ chmod +x pico_setup.sh
$ ./pico_setup.sh

Which had an issue and a ticket opened 
[https://github.com/raspberrypi/pico-sdk/issues/657](url)

LummoxMakes, 2021-11-12:
Hello hello 

After resolving [https://github.com/raspberrypi/pico-feedback/issues/204](url) **this issue remains**

LummoxMakes, 2021-11-12:
Hello hello 

After resolving [https://github.com/raspberrypi/pico-feedback/issues/204](url)

This issue still remains

kilograham, 2021-11-13:
> Thank you for your immediate reply, however as documented on the same forum post you mentioned this solution does not resolve the issue and the reason for the escalation to github

In what way does it not solve the issue? what did you do? what happened?

LummoxMakes, 2021-11-13:
Using your post here 
[https://forums.raspberrypi.com/viewtopic.php?p=1935038&hilit=libusb#p1931172](url)

```
Re: openocd LIBUSB_ERROR_ACCESS
Report this postQuote
Sat Oct 30, 2021 1:37 pm

Assuming your /etc/udev/rules.d/60-openocd.rules contains this (if not you can add it)

Raspberry Pi Picoprobe
ATTRS{idVendor}=="2e8a", ATTRS{idProduct}=="0004", MODE="660", GROUP="plugdev", TAG+="uaccess"

Then you can add your user to the plugdev group
```

Created the required file 
As suggested by fivdi 
`sudo usermod -a -G plugdev pi
`

### Result
**openocd**
```
openocd -f interface/picoprobe.cfg -f target/rp2040.cfg -s tcl
Open On-Chip Debugger 0.11.0-g4f2ae61-dirty (2021-11-12-18:04)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
adapter speed: 5000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: Failed to open or find the device
Error: Can't find a picoprobe device! Please check device connections and permissions.
```

**sudo openocd**
```
sudo openocd -f interface/picoprobe.cfg -f target/rp2040.cfg -s tcl
Open On-Chip Debugger 0.11.0-g4f2ae61-dirty (2021-11-12-18:04)
Licensed under GNU GPL v2
For bug reports, read http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
adapter speed: 5000 kHz
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 5000 kHz
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x10000001
Info : rp2040.core0: hardware has 4 breakpoints, 2 watchpoints
Info : rp2040.core1: hardware has 4 breakpoints, 2 watchpoints
Info : starting gdb server for rp2040.core0 on 3333
Info : Listening on port 3333 for gdb connections
```

LummoxMakes, 2021-11-15:
Working my way through Getting started with RPI Pico 

Trying to use Visual Studio Code with picoprobe by changing launch.json

Before 
`"configFiles": [
                "interface/raspberrypi-swd.cfg",
                "target/rp2040.cfg"
            ]`

After 
`"configFiles": [
                "interface/picoprobe.cfg",
                "target/rp2040.cfg"
            ]`

Terminal window in VSC
````
SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected
Open On-Chip Debugger 0.11.0-g4f2ae61-dirty (2021-11-12-18:04)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
adapter speed: 5000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: Failed to open or find the device
Error: Can't find a picoprobe device! Please check device connections and permissions.

SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...`

```

liamfraser, 2021-11-15:
@LummoxMakes did you reboot after adding your pi user to the plugdev group? You have to log out and back in for a group change to take effect.

LummoxMakes, 2021-11-15:
@liamfraser Thank you very much for your message today 
I'm sorry I have little understanding of Linux as I used it YEARS and I've just forgotten most of it ..

```
pi@raspberrypi:~ $ id
uid=1000(pi) gid=1000(pi) groups=1000(pi),4(adm),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),105(input),109(netdev),117(lpadmin),997(gpio),998(i2c),999(spi)

pi@raspberrypi:~ $ sudo udevadm control --reload-rules && sudo udevadm trigger

pi@raspberrypi:~ $ id
uid=1000(pi) gid=1000(pi) groups=1000(pi),4(adm),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),105(input),109(netdev),117(lpadmin),997(gpio),998(i2c),999(spi)
```

Reboot ..

LummoxMakes, 2021-11-16:
Hello hello 

I have spent another evening and failed to make progress so it's time for me to close the issue and go do something else while someone comes up with a fix

Thank you all for your help

lurch, 2021-11-16:
I think that if the issue is closed, it'll get forgotten about...

LummoxMakes, 2021-11-16:
Hello hello 

@lurch This is my first time using github and I'm new to the RPi 400, RPI OS and the pico so competence is not on my side. After going through the loop four times with clean builds on erased SD cards using the RPi OS and the SDK script following the latest version of Getting Started PDF to the letter I get same result. 

Due to my inexperience **I must be doing something stupid** as no one else has made a comment here and the thread on the forum has gone silent...

LummoxMakes, 2021-11-21:
**I confess to stupidity, bring forth your coffee mugs and biscuit requests and they shall be fulfilled!!**

Sincere apologise for the time wasted on this

**For reference**  : /etc/udev/rules.d/60-openocd.rules 

Raspberry Pi Picoprobe
ATTRS{idVendor}=="2e8a", ATTRS{idProduct}=="0004", **MODE='660'**, GROUP="plugdev", TAG+="uaccess"

Rather than 

Raspberry Pi Picoprobe
ATTRS{idVendor}=="2e8a", ATTRS{idProduct}=="0004", **MODE="660"**, GROUP="plugdev", TAG+="uaccess"

closed 2021-11-21 by LummoxMakes