raspberrypi/pico-feedback #206: Problem with sdk 1.3 and C++ error in ~/pico/pico-sdk/src/rp2_common/pico_platform/include/pico/platform.h

glimpit, 2021-11-16:
After updating to pico sdk 1.3 my C++ project failed to build. On investigation I found the problem was with the following lines in
~/pico/pico-sdk/src/rp2_common/pico_platform/include/pico/platform.h

#if defined(__GNUC__) && __GNUC__ <= 7
#define __force_inline inline __always_inline
#else
#define __force_inline __always_inline
#endif

Changing this to
#define __force_inline __always_inline

fixed the problem.
Using the project generator to create a new vscode C++ project also show up this problem as the generated C++ project does not build.

lurch, 2021-11-16:
Duplicate of https://github.com/raspberrypi/pico-sdk/issues/659 ?

aallan, 2021-11-16:
Quite possibly!

glimpit, 2021-11-16:
I agree it is a duplicate of #659 and has been fixed on #650

closed 2021-11-16 by glimpit