raspberrypi/pico-feedback #207: Programming external QSPI flash via SWD

dima91, 2021-11-19:
I want to reprogram the external flash of Pico via SWD line. This feature must be implemented in a different microcontroller, not from a Linux computer so the OpenOCD library cannot be used.

I already wrote a C library that is able to read and write in RAM and in core registers via SWD and I'm able to put the Cortex-M0+ into halt state.

I read many tutorials and documentation and those that are useful are:

    http://markding.github.io/swd_programing_sram/
    https://www.silabs.com/documents/public/application-notes/an0062.pdf
    https://community.silabs.com/s/article/how-to-program-internal-sram-over-swd?language=en_US

but unfortunately they aren't clear enough.

Can you explain the sequence of commands that I should send to RP2040 to correctly upload the new firmware?
Should I use bootrom functions to implement this operation? How can I call them?

Thanks for your help!

lurch, 2021-11-19:
These all sound like questions for @kilograham

kilograham, 2021-11-19:
I mean the bootrom functions are documented in the RP2040 datasheet.

You can look at how openocd does it https://github.com/raspberrypi/openocd/blob/df76ec7edee9ebb9be86e4cff7479da642b0e8df/src/flash/nor/rp2040.c

closed 2021-11-22 by aallan