raspberrypi/pico-feedback #208: DMA: RING_SIZE and reading WRITE_ADDR

jannic, 2021-11-25:
I'm reading strange values from WRITE_ADDR while using DMA with RING_SIZE > 1.

According to the documentation, RING_SIZE works like a bit mask on the bits to be written while incrementing the address.
So, e.g. with a RING_SIZE of 4 I expect a sequence like 0x2003fb00, 0x2003fb04, 0x2003fb08, 0x2003fb0c, 0x2003fb00, ....
But if I actually read back the value from the WRITE_ADDR register, I sometimes get values slightly before the expected range, in the given example values like 0x2003faf4, 0x2003faf8, 0x2003fafc.

The DMA itself seems to work as expected (data is written to the expected memory rang, and the memory addresses outside the expected range are not affected).

Is there something else to be considered when reading the WRITE_ADDR register?

(Already asked this question at https://forums.raspberrypi.com/viewtopic.php?t=323813, but didn't get a reply.)

Wren6991, 2021-11-25:
Oh -- looks like this is a hardware bug.

When you read from `WRITE_ADDR` or `READ_ADDR`, the DMA subtracts the number of in-flight transfers times the transfer size from the actual live register value, to avoid telling you that an address has been read/written when it is actually still in-flight in the DMA's bus pipeline.

This logic does not take nondefault increment modes into account, so gives odd values when `RING_SIZE > 0 && RING_SIZE <=4` or when increment is disabled.

You should still get sensible values by reading `TRANS_COUNT`.

Wren6991, 2021-11-25:
Thanks for reporting this, I'll add an errata entry.

jannic, 2021-11-25:
Thanks for the quick answer. One remark regarding `RING_SIZE > 0 && RING_SIZE <=4`: I'm quite sure I saw the same behavior with larger RING_SIZE (IIRC 13, can check in case it matters).

Wren6991, 2021-11-25:
> Thanks for the quick answer. One remark regarding `RING_SIZE > 0 && RING_SIZE <=4`: I'm quite sure I saw the same behavior with larger RING_SIZE (IIRC 13, can check in case it matters).

Yes, that was wrong, I did edit my comment once I realised!

jannic, 2021-11-25:
Oh - you already edited your answer. :-)

Wren6991, 2021-11-25:
[image]

This will appear in the next docs release. Congratulations on the first DMA erratum

closed 2021-11-25 by aallan