raspberrypi/pico-feedback #210: Documentation request: Example downstream library that optionally supports pico-sdk

MattMills, 2021-12-07:
I am writing a C library for an i2c accelerometer, and I'd like to support multiple hardware platforms (specifically RP2040 and ESP32's ESP-IDF). 

Is there any chance you could provide an example of a downstream library project that optionally supports pico-sdk?

MrYsLab, 2021-12-08:
I am not sure what you mean by a downstream library, but I developed a [project ](https://mryslab.github.io/telemetrix-rpi-pico/) implemented in pico-SDK that supports i2c, that perhaps might be useful to you.  It implements [an i2c interface](https://mryslab.github.io/telemetrix-rpi-pico/i2c/) that supports two i2c ports. f you are familiar with Arduino Firmata, this project is similar. There is a Python client that communicates with a fixed pico resident application over serial USB.

MattMills, 2021-12-08:
Thanks,

I mostly just meant the cmake magic that will include the pico-sdk if it's requisite for the build target, this: https://github.com/2bn-dev/kxtj3/blob/master/CMakeLists.txt is what I came up with, and it seems like it will work, but IMO it'd be nice to have a template/example of the "pico" way to add a library that depends on the SDK, without having the SDK take over the entire project (IE, if my build is for Arduino or ESP32, I wouldn't want to have to include the Pico SDK).

MrYsLab, 2021-12-08:
Why not use earlephilhower's Arduino Core library instead of the C-SDK? You can then use the Arduino IDE to do your builds and avoid all of the cmake nonsense. I used the SDK in my project because the earlephilhower library was not available until after I completed my code.

There are advantages to using the native SDK's, but using an Arduino Core makes building simpler. :-)

aallan, 2021-12-08:
This seems to be more of a discussion for the [forums](https://www.raspberrypi.org/forums/), so I’m closing this issue.

closed 2021-12-08 by aallan

MattMills, 2021-12-08:
So to clarify, There isn't ever going to be an example library included with the pico-sdk?

lurch, 2021-12-08:
There are plenty of [examples here](https://github.com/raspberrypi/pico-examples), but I'm not sure if "how to support non-RP2040 platforms" falls within the scope of pico-sdk ? :man_shrugging:

MattMills, 2021-12-09:
I'd say it's more how to gracefully incorporate the pico-sdk cmake
ecosystem into a library designed to support multiple boards.

Building an standard application with pico-sdk is not a difficult task, and
I'd appreciate less input from the peanut gallery on how to do things I'm
not asking about.

MattMills, 2021-12-09:
@kilograham Thanks, that clarifies things a bit... Personally I feel that a template repo that demonstrates a library build that shows a native C lib, Arduino lib, and python lib from the same code along with hardware compatibility with pico-sdk and esp-idf (and other hardware where applicable) would be a good way to promote improved compatibility with existing libraries.

I may do that myself, but I'm not at all familiar with CMake, so my solution would likely just be whatever works rather than a good solution that actually takes advantage of the benefits of CMake.

aallan, 2021-12-09:
> Personally I feel that a template repo that demonstrates a library build that shows a native C lib, Arduino lib, and python lib from the same code along with hardware compatibility with pico-sdk and [...]

Sounds like something useful. But we're a small team, and it's very unlikely there will be anyone able to work on this. Sorry.

ricardomatias, 2023-05-21:
@aallan any chances of providing some pointers or a CMakeLists file here? It would be a great help to know how someone just creates an isolated helper library for the pico which relies on a main project importing it as dependency to initialize the sdk.