raspberrypi/pico-feedback #211: cannot open source file pico/stdlib.h or hardware/spi.h

GitbyGit, 2021-12-08:
I am trying to set up a new project folder for an SPI communication example, but I am running into this error of the two libraries. In the first image you'll notice my file set-up. The second picture is my M file set-up as well. the .vscode sets to SDK path which is correct.

[image]

[image]

[image]

lurch, 2021-12-08:
Does the project actually build? I suspect that might just be an "editor warning" rather than a "build warning"?
IIRC there _is_ a way to fix that by telling the editor about your include-paths - perhaps @JamesH65 remembers how?
Alternatively, things might work more smoothly if you use https://github.com/raspberrypi/pico-project-generator ? :man_shrugging:

GitbyGit, 2021-12-08:
Here is the output when building the project folder.

[image]

aallan, 2021-12-08:
> Here is the output when building the project folder…

These are warnings from VSCode, not the compiler. While the Raspberry Pi Pico SDK is compatible with VSCode, it is in fact our recommended IDE, we aren't really in a place to provide support for VSCode-specific issues. This is the sort of question you should be asking on the [forums](https://www.raspberrypi.org/forums/), so I’m closing this issue. Sorry.

closed 2021-12-08 by aallan

brianlmerritt, 2023-12-29:
The issues of squiggly lines are fixed in projects generated by pico-project-generator, so I am adding here in case someone else stumbles on this.

In the file `c_cpp_properties.json` add the following content

```
{
    "configurations": [
      {
        "name": "Linux",
        "includePath": [
          "${workspaceFolder}/**",
          "${env:PICO_SDK_PATH}/**"
        ],
        "defines": [],
        "compilerPath": "/usr/bin/arm-none-eabi-gcc",
        "cStandard": "gnu17",
        "cppStandard": "gnu++14",
        "intelliSenseMode": "linux-gcc-arm",
        "configurationProvider" : "ms-vscode.cmake-tools"
      }
    ],
    "version": 4
  
}
```