raspberrypi/pico-feedback #212: getting-started-with-pico.pdf : 4.4 See "Hello World" USB output. minicom needs sudo on desktop

Sciss, 2021-12-08:
When executing `$ minicom -b 115200 -o -D /dev/ttyACM0` on a desktop linux, nothing happens and `minicom` returns after one second, without any explanation or error. It turns out you need to run it as `sudo`. To be sure, the section assumes you connect from a Pi 4B, but I think it would be good to include this information as an additional "(i) NOTE" box, if one wants to run from another computer (as the previous sections, like 3.2.1, also work on desktop).

Sciss, 2021-12-08:
Perhaps a better way would be to point out the need to be in a particular group? C.f. https://github.com/johnwalicki/RaspPi-Pico-Examples-Fedora/issues/1

rewolff, 2022-03-14:
@Sciss The "without any explanation" is a minicom bug. In the interest of preventing others scratching their head at what's going on reporting that to the minicom developers would be nice. 

I'd say there are serveral ways to fix it. The question is: Would this "getting started" document be the rigth place to explain them all? 

***Note: You need permissions on the serial port. The quick and dirty way is to precede the minicom command with "sudo". Better ways include adding your user to the group of users allowed to access the serial port device. Other ways include configuring udev to grant access  to everybody. What is the best choice for your system cannot be dscussed here.**

aallan, 2022-03-14:
> the "getting started" book is intended to be fairly self-contained

This

rewolff, 2022-03-14:
OK. Agreed... Sounds like a reasonable argument. 

The thing is... People might install something else than raspberrypi_os and have  a different configuration. So then what do you recommend that people do that "always works". 

I'd say that having the user be in group dialout and the device coming up in group "dialout" as on raspberrypi_os would be a reasonable solution. But I just checked (I actually thought I had a similar permission problem in my own project, but that ended up being something else), and the pi user is standard in that group. So then... How did @Sciss end up not being in that group. Added an account for him/herself? Without adding that account to all the groups that pi user is part of? Happened to me before. 

If the user is on Ubuntu, you'd normally be asked the username at install time (not sure if that's the case on pi, but whatever). and that default user can use sudo, but is not in the group "dialout" (even though the device WILL come up with group rw permissions for dialout). 

So even though my two examples would be solved by "make sure that your login user is in the group dialout", it is possible that there are other failure modes that we haven't thought about right now. How far can you debug a problem you're not having at the moment that might occur on a user's machine with SOME configuration ?

aallan, 2022-03-14:
> People might install something else than raspberrypi_os and have a different configuration.

The Getting Started book assumes that the user is running Raspberry Pi OS. 

There are chapters supporting Windows and macOS, but beyond that it's assumed that you're running Raspberry Pi OS. Considering the number of different Linux distributions it's going to be impossible to support them all, so we're not picking favourites. We do make a very good Linux-based computer and recommend it, and our OS, as the development environment for Pico and RP2040.

lurch, 2022-03-14:
> So then... How did @Sciss end up not being in that group.

He talks about "desktop linux" so I assume he's talking about a generic Linux distro running on a x86 PC. And there's obviously such a huge variety of distros, each with different methods of configuration, that we've deliberately not gone into the effort of documenting those in detail, as it would make the getting-started guide too long and unwieldy.

A Raspberry Pi (and SD card, etc.) is often cheaper than just the programmer-device or IDE for other microcontrollers :wink:

Sciss, 2022-06-01:
> He talks about "desktop linux" so I assume he's talking about a generic Linux distro running on a x86 PC.

Yes, it's standard Debian stable, with standard user.

lurch, 2022-06-08:
I'm going to close this. As already mentioned, the instructions work out-of-the-box on Raspberry Pi OS, and there are so many alternative ways of fixing this on other Linux platforms (using sudo, or using groups, or using udev, or...) that IMHO it feels beyond the scope of the getting-started guide.

closed 2022-06-08 by lurch

lurch, 2022-06-08:
But then, in how many different places in that document would we need to add a "on other systems check the permissions if you're having trouble" note? :confused:   (Also, I think "check the permissions" is too vague compared to the rest of the "do this, then do that" style of language used in that document)