raspberrypi/pico-feedback #214: Loading firmware into RP2040 SRAM via SWD

dima91, 2021-12-29:
I have to upload a new firmware in the Raspbery PI PICO external flash via SWD from a different microcontroller (an ESP32) connected via UART and SWD lines (I asked [here a previous question](https://github.com/raspberrypi/pico-feedback/issues/207)).  
I decided to firstly upload a firmware into PICO SRAM and, secondly, send the firmware binaries via UART to the just uploaded firmware which, in turn, writes the firmware in FLASH memory.  
The firmware that I try to load into SRAM is wrote with the C/C++ PICO SDK and it is compiled with `PICO_NO_FLASH` option enabled by adding `pico_set_binary_type(TARGET no_flash)` to CMakeLists.txt file.

At the moment, I wrote a custom SWD library to write the firmware in PICO SRAM, start and stop cores and write core registers.

To perfom such operation, I followed these useful tutorials (and many others, but they are the most relevant)
+ http://markding.github.io/swd_programing_sram
+ https://community.silabs.com/s/article/how-to-program-internal-sram-over-swd

The problem appears when I try to restart the execution of RP core which doesn't execute the just loaded firmware but it remains stuck.

More in details, to **halt the core**, I write:
+ 0xA05F0003 to DHCSR register
+ 0x1 to DEMCR register
+ 0x05FA0004 to AIRCR register

after that operations the DHCSR register contains value 0x3030003.  
Then I write the new firmware to SRAM starting from address 0x20000000 (and checking wrote values) in chunck o 512 bytes.  
Finally I uptdate:
+ VTOR register value with 0x20000000 (the base address from which I wrote the firmware)
+ PC core register value (having index 15) with the second word of incoming firmware
+ SP core register value (having index 13) with the first word of incoming firmware

and I **reset the core** simply writing 0xA05F0000 to DHCSR register.

However the RP remains stopped.

I also tried to change VTOR value to 0x20000100 or to reset the RP at the end of the process by writing 0x05FA0004 to AIRCR but the result is the same, in the first case, or the firmware is loaded from FLASH memory, in the second case.

Is it a problem of VTOR, PC or SP addresses?  
I already tested separately the functions to stop and restart the core and they works correctly.

Can you help me?

aallan, 2022-01-04:
This is the sort of question you should be asking on the [forums](https://www.raspberrypi.org/forums/), so I’m closing this issue. But please follow up with further comments if you have specific questions about the documentation.

closed 2022-01-04 by aallan

lurch, 2022-01-04:
You might want to have a look at how https://github.com/raspberrypi/openocd does things? :shrug:

dima91, 2022-01-10:
I saw at the openocd sources only for that concerning the programming of external flash.
Thank you for the tip! 😊