raspberrypi/pico-feedback #218: Broken links to static.docs.arm.com

jannic, 2022-02-02:
The RP2040 datasheet contains two links which seem to be dead.
On page 44, section "2.4.1.3. ARM architecture", there are links to https://static.docs.arm.com/ddi0419/d/DDI0419D_armv6m_arm.pdf and https://static.docs.arm.com/ddi0484/c/DDI0484C_cortex_m0p_r0p1_trm.pdf, but static.docs.arm.com doesn't resolve.
If this isn't just a temporary failure, those links should be updated.

It looks like those documents are still available at https://developer.arm.com/documentation/ddi0419/latest/ and https://developer.arm.com/documentation/ddi0484/latest/

Other links in the datasheet already point to developer.arm.com.

lurch, 2022-02-02:
Thanks, looks like Arm must have rearranged their documentation site since we wrote the datasheet. Shame they didn't put any [redirects](https://github.com/raspberrypi/documentation/pull/2018) in place :wink: 
We'll make sure to fix this in the next datasheet release. (and BTW I think you meant "On page 64")

jannic, 2022-02-02:
Yes, correct, page 64. Sorry.

aallan, 2022-02-07:
Checked with Nick, he says… 👍 

Fixed on `develop`, will be in the next documentation release.

closed 2022-02-07 by aallan