raspberrypi/pico-feedback #221: Clarification on the maximum number of insturctions allowed per PIO State Machine

spit4520, 2022-02-15:
Hi,

The documentation and the compiler do both state that you can have a maximum of 32 instructions per PIO, but it wasn't/isn't clear that its explicitly per state machine. 

What would make this easier: 

Have some form of statement directly on the RP2040 datasheet/pico-sdk that states each state machine has its own 32 instruction maximum (even though they are apparently shared between the different state machines)

I would like to also suggest a linter/compiler macro that will yell and warn that you can have 32 instructions per state machine, but per PIO. The current warning states you can havea maximum of 32 per PIO, which lead to a rabbit hole on finding the answer and eventually this issue

Thanks

lurch, 2022-02-15:
Huh? Section "3.2. Programmer’s Model" of https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf says "The four state machines execute from a shared instruction memory.". And see also section "3.2.7. Interactions Between State Machines" where it says "The instruction memory is implemented as a 1-write 4-read register file, so all four state machines can read an
instruction on the same cycle, without stalling.".

rewolff, 2022-03-14:
Lets be clear: Each of the 4 statemachines of a PIO can use 0-32 instructions (zero: disabled) But the sum of the four may not exceed the 32 they share in the PIO.

lurch, 2022-03-14:
I think talking about "sum of the four" isn't helpful, as you could e.g. have a 20-instruction PIO program, with 3 PIO statemachines all running the same program (numbers picked arbitrarily), just with different per-statemachine pin-configs. And "sum of the four" in that case would be 60.

aallan, 2023-09-05:
Seems to be resolved, but not closed. So closing now.

closed 2023-09-05 by aallan