raspberrypi/pico-feedback #222: Collection of lacking information im RP2040 manual

MarcVonWindscooting, 2022-02-23:
While working with the manual on my own build environment I noticed, that essential information is lacking and some things plain wrong. I recorded these for some time. Here they are:

1. (2.15. CLOCKS) Does (only) PERI CLK really NOT have a divider ? The register documentation is missing completely. Figure 28: clk_sys also drives PWM - this is missing!
2. (4.1. USB) SIE_STATUS . DATA_SEQ_ERROR cannot be cleared by writing a 1 (this is how I understand 'WC') but by REALLY clearing it, e.g. through CLR alias. SIE_STATUS . SPEED is 'WC' too? That seems odd to me. Completely lacking: the meaning of USB_MUXING and USB_PWR registers. At least I could not guess the meaning of the bits.
3. (4.5. PWM) no clear mention of clock source. 4.5.2.4 calculations suggests SYS CLK. 4.5.1 is where one searches for the information.

liamfraser, 2022-02-24:
Hi, thanks for the feedback. You are right on all points.

- clk_peri has no divider
- figure 28 could be updated to show that clk_sys drives pwm (this info could be gathered from the WAKE_EN/SLEEP_EN registers but I appreciate it is not obvious). It might be tricky to fit everything in the diagram though.
- Yes the USB status bits are expected to be cleared with the clear alias. The reason things like speed are latched is because they are used as interrupt sources. So you clear the interrupt by clearing the bits
- USB muxing is really for internal testing / fpga. On the real chip you would always pick the "to_phy" option.
- The USB_PWR registers are used to override the signals into the usb controller that can come from gpios: vbus detect etc
- Yes, we should make it clear in the PWM chapter saying the source clock is clk_sys

rewolff, 2022-03-14:
> USB muxing is really for internal testing / fpga.
> On the real chip you would always pick the "to_phy" option.
I would suggest to document them as: "reserved, leave at reset state" if possible. I appreciate the honesty that there really IS a register there. For us mortals out in the real world (WITH real chips, WITHOUT the FPGA-source) "reserved" is sufficient. 

On the other hand.... documenting the debug features may help someone come up with a clever hack. 

So documenting as 
"Reserved, leave at reset value. These bits were used to mux the USB output during development of the chip."  might be in line with raspberry pi policy. 

Reading between the lines, the chip would have gone out even when PIO wasn't working. Why else would there be SPI, I2C and Uart modules? I'd say a third and fourth PIO woul'dve been about the same area, but more functionality.

LukasGossmann, 2023-06-06:
As MarcVonWindscooting already noticed correctly clk_peri does not have a divider so it should be removed from the diagram. There is a related issue for the SDK where i also noticed that.

Copy paste of the info from there:
```
Here is what needs to be changed:
1: On page 181 the Figure 28 needs to have its box with the division sign removed in the path of clk_peri.
(build-date: 2023-03-02, build-version: ae3b121-clean)
```