raspberrypi/pico-feedback #224: RP2040 DMA IRQ does fire when channel is aborted

essele, 2022-03-03:
Hi,

The documentation clearly states that aborting a DMA channel does not cause an interrupt, however all of my testing clearly shows it does. I assume this is a documentation error rather than a chip errata?

Some sample code to demonstrate it:

````
#include <stdio.h>
#include <string.h>
#include "pico.h"
#include "pico/stdlib.h"
#include "hardware/dma.h"
#include "hardware/irq.h"

int                 dma_chan;

uint8_t		    mem1[32768];
uint8_t		    mem2[32768];

void dma_isr() {
    dma_channel_hw_t *hw = dma_channel_hw_addr(dma_chan);
    dma_channel_acknowledge_irq0(dma_chan);

    uint32_t bytes = (uint32_t)hw->write_addr - (uint32_t)mem2;
    printf("DMA STOPPED at bytes=%d\r\n", bytes);
}

int main() {
    stdio_init_all();

    sleep_ms(1000);
    printf("Starting.\r\n");

    dma_channel_config dma_ch_config;

    dma_chan = dma_claim_unused_channel(true);
    dma_ch_config = dma_channel_get_default_config(dma_chan);

    irq_set_exclusive_handler(DMA_IRQ_0, dma_isr);
    irq_set_enabled(DMA_IRQ_0, true);
    dma_irqn_set_channel_enabled(0, dma_chan, true);

    channel_config_set_read_increment(&dma_ch_config, true);
    channel_config_set_write_increment(&dma_ch_config, true);
    channel_config_set_transfer_data_size(&dma_ch_config, DMA_SIZE_8);

    dma_channel_configure(dma_chan, &dma_ch_config, mem2, mem1, 32768, false);

    dma_channel_start(dma_chan);

    sleep_us(50);
    dma_channel_abort(dma_chan);

    while(1);
}
````

aallan, 2022-03-03:
Ping @liamfraser

Wren6991, 2022-03-03:
> The documentation clearly states that aborting a DMA channel does not cause an interrupt

That's right. The documentation matches the hardware in this case, too. If there is a situation where that doesn't hold, then that *is* an erratum, not a documentation error.

> however all of my testing clearly shows it does.

The only thing your software shows, as far as I can tell, is that your channel completes before the abort takes place. Seems fairly plausible since you are running from flash.

Wren6991, 2022-03-04:
So

- Yes, this is a hardware erratum, thank you for raising it
- The specified behaviour was: abort does not cause an IRQ in any circumstances
- The actual behaviour is: abort *does* cause an IRQ if there are in-flight transfers (i.e. read but not yet written) on that channel when the transfer is aborted.
- This means aborting a channel that has stalled on an inactive DREQ will predictably not cause an IRQ, but aborting a channel that is not in a stalled state is unpredictable

We should be able to find a suitable software workaround and add it to the abort function in the SDK, and add an errata entry in the datasheet.

Wren6991, 2022-03-04:
> The actual behaviour is: abort does cause an IRQ if there are in-flight transfers (i.e. read but not yet written) on that channel when the transfer is aborted.

...and this is a consequence of the ABORT status clearing before those transfers have finished flushing through, so that bit can not be reliably polled, and CTRL.BUSY should be polled instead.

Wren6991, 2022-03-04:
I've added a workaround to the SDK which seems to stop your testcase from triggering an IRQ. Could you try it out with your original code please?

Errata entry will appear in the next docs release:

There is also a pointer to this in the main body of the datasheet:

[image]

essele, 2022-03-05:
Tested with my original code and it works ok -- it's actually what I did as a workaround originally (although I was still waiting on ABORT rather than BUSY.)

Thanks for the quick response.

lurch, 2022-06-16:
Thanks, will be fixed in the next PDF release of the databooks.

closed 2022-06-16 by lurch