raspberrypi/pico-feedback #226: PADS_BANK0: VOLTAGE_SELECT Register

rewolff, 2022-03-18:
In the description for the PADS_BANK0: VOLTAGE_SELECT Register: 

`0x1 → Set voltage to 1.8V (DVDD <= 1V8)`

should probably read 

`0x1 → Set voltage to 1.8V (DVDD < 2V5)`

dp111, 2022-03-18:
Warning from datasheet:

"Using IOVDD voltages greater than 1.8V, with the input thresholds set for
1.8V may result in damage to the chip"

closed 2022-03-18 by aallan

rewolff, 2022-03-18:
Then the line above should read > 1.8V 

Although 1.8 and 2.5 are common voltages, it should be documented what I should do with 2.0 and 2.2V  VDD, right ?

dp111, 2022-03-18:
More from the datasheet

"Both the output signal level and acceptable input signal level at the pad
are determined by the digital IO supply (IOVDD). IOVDD can be any nominal
voltage between 1.8V and 3.3V, but to meet specification when powered at
1.8V, the pad input thresholds must be adjusted by writing a 1 to the pad
VOLTAGE_SELECT  registers. By default the pad input thresholds are valid
for an IOVDD voltage between 2.5V and 3.3V. Using a voltage of 1.8V with
the default input thresholds is a safe operating mode, though it will
result in input thresholds that don’t meet specification."

rewolff, 2022-03-21:
OK. I got it!  The chip cannot be used between 1.8V and 2.5V because there is no acceptable value for the PADS_BANK0 bit. 

Thanks for clarifying that. 
But then doesn't ".. any nominal voltage between 1.8V and 3.3V" need some clarification?

dp111, 2022-03-21:
It can be used between 1.8v and 2.5v. Just the input logic levels will be
different from what might be expected.

rewolff, 2022-03-21:
IMHO there is still not enough information in the datasheet to make an informed decision. As written the datasheet tells you to use 0 for <= 1.8V . and 1 for => 2.5V. There is no recommended value for 1.8...2.5V (between 0 and 1 maybe? I thought that register was binary!). 

If it is 0 for dvdd <= 1.8V and 1 for dvdd > 1.8V (*) then maybe a note to clarify things: "threshold levels may be wrong for voltages between 1.8 and 2.5V. " 

(*) tricky... what to do if your 1.8V powersupply is just a tad off and produces 1.9V. Now what? I suspect 1.8V Nominal (up to 1.98V) is OK.

rewolff, 2022-03-21:
I don't care if the value in the register needs to be 0 or 1. I'll look it up if I need it. (the default is OK for 3.3V) 

I understand what you want to say. 

I think that PADS_BANK0: VOLTAGE_SELECT should explain the situation better or refer to the other sections you  list. 

Undefined behaviour occurs with DVDD voltages between 1.8V and 2.5V, and that's precisely what the description implies, so technically that dexcription is precisely correct.

dp111, 2022-03-21:
" I don't care if the value in the register needs to be 0 or 1. I'll look
it up if I need it. (the default is OK for 3.3V)"

I do care as this may pop up in an internet search and lead people down the
wrong path.

With all registers if you need more detail you need to refer to the
relevant section of. Searching the datasheet for " VOLTAGE_SELECT" gives
the relevant information.

dp111, 2022-03-21:
> OK. So lets assume I plan on using 2.0V for the DVDD supply for some
> reason. (e.g. looks possible for the RP2040, and I need precisely that
> voltage elsewhere)
> I then find PADS_BANK0: VOLTAGE_SELECT Register description and it doesn't
> specify what to do for my case. I then search for "VOLTAGE_SELECT" and find
> which again doesn't specify any behaviour for voltages between 1.8 and
> 2.5V. Reading between the lines it is "unspecified", but looks very much
> like a datasheet-typo.
> as using "1" in the register at voltages above 1.8V can damage the chip,
> that setting should clearly be reserved for <= 1.8V operation. So by
> definition of a binary bit, the value "0" should be used for dvdd > 1.8V.
> Why are you objecting to writing that in the datasheet for "0" in the
> voltage select bit? "Input voltage levels are nonstandard below 2.5V"? Then
> just add that note there. Much clearer!

I can't see anything wrong with the datasheet. If you want to use the
RP2040 at a voltage that isn't supported you are welcome too. We only
document 1.8V 2.5V 3.3V ( the standard logic levels)  outside that range
you are on your own. For 2.0V just don't set the bit, but including that
case in the datasheet may confuse people who think it might be a support
voltage.