raspberrypi/pico-feedback #227: SPI output enable does not tristate

djfarrell, 2022-03-25:
Using the SPI as a slave with multiple RP2040s sharing the bus.  The documents (for the cell) indicates there is an output enable on the TX line that should be controlled by the select.  I see no means of enabling this (if it really exists).  The MOD implies a possible option, but all this bit seems to do is disable the output shift register, the driver is still there.  I have to manually switch between SPI and GPIO on the pin of un-selected slaves or respin my PCB using drivers of or'ing pins.. Am I missing something?

lurch, 2022-03-25:
ping @liamfraser

Wren6991, 2022-03-28:
Thanks for raising this.

> The documents (for the cell) indicates there is an output enable on the TX line that should be controlled by the select. I see no means of enabling this (if it really exists).

I went and confirmed the hardware implementation, and you are correct:

- On RP2040, the SPI_TX pin function is hardwired to always assert the pad output enable
- The Arm PL022's SSPTXD output enable is not connected; PL022 does not have tristate control of this pin

So yes, software needs to manually switch the output enable when multiple RP2040 SPI slaves are sharing a bus, either by switching GPIO function as you are doing, or by toggling the `oeover` field of the relevant `iobank0.ctrl` register.

I'll fix the documentation to match the hardware behaviour, as this is clearly a serious mismatch -- this will appear in the next docs release. I'll also raise an internal note to review the SSPTXD output enable on future devices which use the PL022. Thanks for the report, and sorry for the trouble.

lurch, 2022-06-16:
Thanks, will be fixed in the next PDF release of the databooks.

closed 2022-06-16 by lurch

alien999999999, 2023-06-03:
> So yes, software needs to manually switch the output enable when multiple RP2040 SPI slaves are sharing a bus, either by switching GPIO function as you are doing, or by toggling the `oeover` field of the relevant `iobank0.ctrl` register.

So, how can i toggle this field or gpio function at the correct timing? considering a fifo is used, and an IRQ is slow to react, it will just be too late? I've given up and put a IC on the pico (slave) TX after this, which floats the output when CSN is high ... and otherwise keeps the output.

Is there really a way to fix this in software? if so, that would be great to be added to doc.

tomtom7899, 2024-10-24:
Hey, does anyone know how to solve this in software?

alien999999999, 2024-12-16:
Would it be possible to have in the firmware to instantly toggle oeover on CSN change? it could be a software option to enable this firmware workaround?

Wren6991, 2024-12-18:
The software workaround for RP2040 is to tristate the bus manually when deselected. If you have sufficient setup on the chip select, you can do this directly in software (e.g. a GPIO interrupt). If you have tighter timing requirements, you can use a PIO state machine to monitor the pin, and pass register values from PIO RX FIFO to a DMA channel set up to control the pad tristate via IO_BANK0_GPIOx_CTRL_OEOVER.

We fixed this in hardware for RP2350, so the SSPTXD output is now tristate-controlled by the peripheral.

alien999999999, 2024-12-20:
Thanks for the reply and fix in next chipset.

I did not realize it's possible to set up a dma to that register, unfortunately, I already have 2 pio programs on my PICO 1's, so that doesn't work for me, but good to know for the future.