raspberrypi/pico-feedback #231: USB Host Controller documentation

alimansfield2016, 2022-04-14:
There are a couple issues I think I've found with the HCD documentation, and a couple of queries regarding the behavior of the PHY.

Firstly, in section 4.1.2.7.2. HOST CONTROLLER - IN, for the control phase, EPx control register is specified at location 0x80, whereas in Table 404 - DPSRAM Layout, this register is placed at 0x100. 
Again similarly EPx buffer control register is given to be at 0x100, where it should be 0x80.

Secondly, there seems to be no information on the behavior of the Host In Data phase if the device NAKs instead of returning data. How is this handled by the hardware?

Thirdly, in Table 404 there are the "Spare" registers. In Host mode, are these completely ignored by the controller? Or could they be used similarly to the Interrupt IN endpoints, but rather as OUT endpoints.

Finally, on a similar note to my third point, Can the "Interrupt" endpoints be used for non-interrupt transfers? If so what is the behavior of this if multiple are enabled/active simultaneously?

lurch, 2022-04-14:
ping @liamfraser (and also @hathach, in case he has any insights)

alimansfield2016, 2022-04-15:
To answer all of my questions here (I am 90% confident, but correct me if i'm wrong)
- The NAK behavior of the USB Host is to block and endpoint until data is provided. After NAK, the controller will wait for NAK_POLL microseconds. The reset value is 16us, would recommend setting it higher.
- The "Spare" registers are defunct in Host mode. The Interrupt endpoints can be used for both IN and OUT transactions by setting the relevant bit in ADDR_ENDPn
- The Interrupt endpoints *should probably* be renamed, since they can be used for any type of endpoint and are not just limited to Interrupts, but I assume that's unlikely. These endpoints are all asynchronous - the controller will deal with them in order and fire the relevant interrupts.
I have managed to get Bulk endpoints working completely and will be pushing my changes into TinyUSB when I get the chance to complete the RP2040s HCD.

hathach, 2022-11-06:
related PR https://github.com/hathach/tinyusb/pull/1434 . "interrupt" endpoints can be indeed used for Bulk (probably not ISO). However I have an question regarding the throughput. It would be great if RPI team could clear the cloud (though I will do more hand-on test to verify this later on), here is what I mentioned in the above PR:

Basically the interrupt and bulk have the same packet layout (pid, data, crc, ack), the difference is that bulk endpoint will continuously transferring data as long as bus frame (1ms) is allowed. For example, we can have 10 packets of 64 bytes in 1ms with Bulk. For interrupt endpoint, it will only transferring 1 packet per interval (could be 1ms, 2ms, 4ms etc..). To be honest, I am not sure if using these "interrupt" endpoint for bulk this way would hurt the throughput or not (need to confirm with rpi team). If it does then our max throughput would be 64 bytes/1ms --> 64KB/s . IMHO, it will likely be the case here. Though it is still better than nothing now. We can merge this as it is and try to figure out a better implementation later e.g similar to https://github.com/hathach/tinyusb/pull/1219 approach