raspberrypi/pico-feedback #232: Errors with micropython build in RPI PDF document

euro-phil, 2022-04-22:
This is following on from a question and (possible / proposed )answer set, I have received.

It would seem a change has been made to the micro python git sources, and this now prevents the example in the PDF 
https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-python-sdk.pdf

My view point is from a lecturing perspective and having students follow these instructions.... ( which currently do not work)

Whilst I can work with these to get an working end result, with the solutions outlined in the following chat(s), it has been suggested that having to revert to using make commands rather than cmake, is not ideal from a teaching perspective. 

the chat is located 
https://forums.raspberrypi.com/viewtopic.php?t=333395&sid=ab26e88600648d9365ca3b9305652b09

Many thanks due to B.Goode and hippy for the valuable perspectives on the issues and functioning solutions

Phil.

lurch, 2022-04-23:
Thank you for bringing this to our attention. I've just looked into it, and the lines in our PDF that say:
```
$ cd micropython
$ git submodule update --init -- lib/pico-sdk lib/tinyusb
```
should instead say:
```
$ cd micropython
$ make -C ports/rp2 submodules
```

This will be fixed when we next do a release of our databooks, although we're not sure when that'll be yet.

dpgeorge, 2022-04-26:
Yes, we recently added a new submodule dependency, mbedtls, to get some crypto functions in the firmware.

We could add a workaround in the makefile to automatically fetch this new submodule, but it would be much nicer if the docs could be updated, as suggested above.

lurch, 2022-04-26:
Thanks for the confirmation @dpgeorge 
I ran through the instructions (in our PDF) recently, and also noticed that it seems like the `build` directory has now been renamed to `build-PICO` ?

lurch, 2022-05-25:
Fixed internally, will be included in the next release of the PDFs.

closed 2022-05-25 by lurch

euro-phil, 2022-05-25:
Hello Andrew,

Many thanks for this, keep up the excellent work.
Kind regards
Phil