raspberrypi/pico-feedback #233: NVIC (timer) interrupt: no preemption without dynamic priority level

b-zee, 2022-04-28:
According to section 2.3.2 in the RP2040 Datasheet:

> a lower-priority interrupt can be preempted by a higher-priority interrupt

> for interrupts with the same dynamic priority level, the lower-numbered IRQ has higher priority

During testing I have not been able to have the `TIMER_IRQ_1` preempted by `TIMER_IRQ_0` (without setting the dynamic level). When setting the dynamic level of `TIMER_IRQ_1` to `3`, it does get interrupted by `TIMER_IRQ_0`, but according to the cited documentation that should not be necessary.

b-zee, 2022-04-28:
After some thinking it seems my mental model isn't exactly right. Although the documentation isn't exactly correctly describing it either.

It seems that the priority in terms of preemption is different from the order in which pending exceptions are activated. I think these are both referred to as 'priorities', while there is a difference. Perhaps someone more enlightened than me can put this in more succinct wording to highlight that my observed behavior is the correct behavior according to the ARMv6-M architecture...

aallan, 2023-03-02:
Does raspberrypi/pico-sdk#245 resolve this issue?