raspberrypi/pico-feedback #234: Documentation for PIO: add clarification of MOV from PINS

Tails86, 2022-05-03:
For the datasheet and SDK documentation under the MOV PIO instruction (currently under 3.4.8 in both), I suggest a note for source of PINS that the word copied from PINS will be ALL pins starting from PINCTRL_IN_BASE set in the config (or whatever exactly is happening because I don't know for sure). 

I had forgotten that the range is not specified for sm_config_set_in_pins. It took me a while to understand why checking X after `MOV X, PINS` was not working as expected, and I needed to first `MOV OSR, PINS` before shifting out the number of bits I desired into X.

Thank you for your time in considering this.

lurch, 2022-05-03:
Does section 3.5.6.2. of https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf answer your question?

Tails86, 2022-05-03:
This is more of a request to add a note in the documentation to remind the reader that the MOV command copies all pin values from the PINS source

Wren6991, 2022-05-04:
I added this note at the bottom to 3.4.8:

[image]

If you feel like this clarifies things, it will appear in the next docs release.

lurch, 2022-05-04:
Thanks Luke. Just for completeness, is a similar comment warranted for `MOV PINS, src`? Or is that not needed because of `PINCTRL_OUT_COUNT`?

(Hah, it's never occurred to me before, but `MOV PINS, PINS` might actually be a useful thing to do if you have different `IN` and `OUT` pin mappings :slightly_smiling_face: )

Wren6991, 2022-05-04:
I'd assumed that was okay, since the out mapping has a count

Tails86, 2022-05-04:
Yes, I think that will help out a lot. Thank you for hearing me out :)