raspberrypi/pico-feedback #238: PIO sideset wait behavior

tmcqueen-materials, 2022-05-15:
First, thanks for the wonderful product and projects. While implementing the high-speed version of I2C in PIO [pio-i2c-hs](https://github.com/tmcqueen-materials/pio-i2c-hs), an ambiguity in the specification came up. Specifically, suppose a single GPIO pin X starts at logic level 0, and then the PIO SM executes a `wait 1, pin X side 1` instruction, where the sideset set pin X to logic level 1. What is the minimum number of cycles that will be taken by this instruction?

In my testing, the answer appears to be [`2`](https://github.com/tmcqueen-materials/pio-i2c-hs/blob/main/i2c.pio#L63), with the wait sampling the pin before the sideset brings it to logic level 1, but I could not find anything in the Pico specification about whether this is the intended behavior, the actual behavior in all cases (but not guaranteed to behave similarly in the future), or a fluke of my specific testing setup. Are there any specification guarantees (or non-guarantees) about such an instruction that simultaneously samples and sidesets the same GPIO pin?

Thank you!

PS: I am happy to contribute the above repo to pico-examples if you would accept such a pull request.

Wren6991, 2022-05-16:
We don't have any special logic for when the side-set pin matches the wait pin. The wait is based on the input value observed at the input to the pad, and the side-set controls the output (-enable) to the pad. The only reason these two logic paths interact at all is the operation of the pad (which is inherently analogue).

* Cycle 0: WAIT is blocked, but side-set executes. Side-set's pin update is registered into the IO mapping logic at the end of the cycle.
* Cycle 1: the pin update propagates through the IO mapping logic to the pad, charges the capacitance of the external trace, and (possibly) sets the pad input by the end of the cycle, depending on external capacitance and clk_sys frequency

So yes, you are measuring the round-trip time of output to input, which is around 4 cycles when the input synchronisers are enabled, and ~2 cycles when they are bypassed.

tmcqueen-materials, 2022-05-16:
@Wren6991 Thank you very much for the clarification!

closed 2022-05-16 by tmcqueen-materials