raspberrypi/pico-feedback #239: HW feature request: UART Transmit Complete interrupt

eskimo-software, 2022-05-23:
It's often necessary to detect when the last byte of a transmission has fully completed - that is, when the tx shift register is empty (and stop bits have been transmitted) and there's nothing in the tx FIFO to start a new byte transmission.

This would appear to correspond to the BUSY flag in UARTFR transitioning to 0.

Generating a Transmit Complete interrupt on this transition would simplify detection and handling of this case.

lurch, 2022-05-23:
Tangentially related to the discussion in https://github.com/raspberrypi/pico-sdk/issues/548 particularly "The PL011 doesn't seem to have the necessary status bits for you to poll in order to wait for the current TX/RX characters to complete like it asks you to do. This code just adds a 15 baud period busy wait after disabling the UART, to make sure in-progress characters have finished."

The PL011 reference manual is [here](https://developer.arm.com/documentation/ddi0183/).

lurch, 2025-07-09:
I'm going to close this as a duplicate of https://github.com/raspberrypi/pico-sdk/issues/1274  (see the "workaround" in https://github.com/raspberrypi/pico-sdk/pull/1709/files )