raspberrypi/pico-feedback #246: Trivial errors in RP2040 datasheet

arg08, 2022-06-17:
(based on version build-date: 2021-11-04 build-version: 150df05-clean)

2.4.8, M0PLUS: CCR Register (P87 of the PDF): two of the fields are documented as "always reads as 1" but the reset value is listed as 0x0

M0PLUS: CCR Register (P98 of the PDF): the CAUTION box at the end of the section has an incomplete sentence saying "Starting the channel".  I'm not quite sure what that is trying to say - the first sentence makes sense on its own, so this was either intended to make a different point or is just junk from editing

JamesH65, 2022-06-17:
I think the second of those is fixed in the latest release (actually today), but the first is still present/inconsistent,

lurch, 2024-05-09:
@nathan-contino I just stumbled across an old email from Nick Francis where he says "Yep this should have reset value read as 1 for those 2 bits."