raspberrypi/pico-feedback #251: A user's view

petercwallis, 2022-07-03:
I am no CMake expert, and indeed don't know what files are needed or what goes in them.  I can, of course, push the buttons which is I guess what most people do.  

The use-case for most people is however not to compile all the examples and make sure they still work, but to compile one example and figure out how it was done.  The documentation is good for what is there but  PLEASE can someone restructure the two examples and documentation so that the instruction 
`cmake ..`
doesn't reference a file that is 54 lines long with two includes, and then have local cmake files for each example that also have includes.

The cmake file for blink.c should be stand alone so the instructions can match the use-case. Sure including stuff rather that duplicating it does make maintaining it easier, but it doesn't help us learners.  I think something like this would help:
"In order to blink the LED on the Pico board (GPIO pin25) you will need to make a uf2 file, then copy it over to 2040's memory via a usb cable. To do that, make an empty working directory (e.g. ~/build) and copy blink.c and cmake.txt from ??? into the directory.  You may also like to make sure that when you installed the pico-sdk, it set up some environment variables including PICO_SDK_PATH which should be pointing to the "..../pico-sdk" directory. Change to the build directory and type cmake. This will: 1) use gcc to compile the code, 2) use ln to... 

If the code does not compile, you will need to...

If the make process cannot find ... you will need to...

As a new learner I hope my reporting of the issues I face is helpful.

VeloSteve, 2023-03-04:
Another new Pico learner here in a similar situation.  I took the advice of using Chapter 8 from Getting Started, but I used the code from the hello_multicore example.  cmake .. runs with no errors, but make fails with a no such file or directory error on pico/multicore.h.

There seems to be no such file on GitHub in raspberrypi/pico-examples, and I don't see it in pico-sdk.  This is the sort of thing that drives new users crazy.  Examples should be easy starting points from which to build, not a barrier.  Maybe multicore.h gets built somewhere in the  stack of 3 levels of CMakeLists files, but why make such dependencies in an example?

I'm really not asking for how to get this particular example running.  I'm asking for examples which are easy to run.

p.s. I am new to Pico and not up to date on current build tools, but I have programmed in C for years, and run many a make file.  I don't mind dependencies if they are well defined.

lurch, 2023-03-05:
If you look at https://github.com/raspberrypi/pico-examples/blob/master/multicore/hello_multicore/CMakeLists.txt you can see that it adds `pico_multicore` to the `target_link_libraries` - I believe you just need to add this to your own standalone `CMakeLists.txt` (that you've created following the instructions in Chapter 8) and then things should work okay? You can read more about this in https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf

`pico/multicore.h` *is* in pico-sdk ([here](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_multicore/include/pico/multicore.h) if you're curious) and it's functions are documented [here](https://www.raspberrypi.com/documentation/pico-sdk/high_level.html#pico_multicore).

nathan-contino, 2024-11-07:
We've since restructured the Getting Started guide significantly. Closing this feedback since those changes _should_ address this problem. If this is still a problem, feel free to create another issue with a specific recommendation and I'm happy to implement it.

closed 2024-11-07 by nathan-contino

lurch, 2024-11-07:
Also, we've released the new-and-improved VS Code extension since this issue was first opened :slightly_smiling_face: 
https://www.raspberrypi.com/news/get-started-with-raspberry-pi-pico-series-and-vs-code/