raspberrypi/pico-feedback #253: "Which hardware am I running on" doesn't work for usb stdio

lbt, 2022-07-03:
In https://datasheets.raspberrypi.com/picow/connecting-to-the-internet-with-pico-w.pdf the section 2.4 example is compiled with ```pico_enable_stdio_usb(test 1)``` however this does not work in practice.

I suspect this is because the usb initialisation happens too slowly and the data is not sent before the cpu halts at the end of main()

Replacing the plain print with this snippet works for those of us using a desktop machine with ttyACM0 over usb to see the output.

```
	while (true) {
		printf("GP25 value: %i\n", value);
		sleep_ms(1000);
	}
```

lurch, 2022-07-04:
This is a "common problem" for examples which use printf with a USB-stdio and which don't loop forever.
Have a look at `PICO_STDIO_USB_CONNECT_WAIT_TIMEOUT_MS` in https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf for one possible workaround.

lbt, 2022-07-04:
OK - although it's a bit of a rabbit hole trying to fix all possible issues, maybe it's worth having code that works for this not-that-rare class of user given that this code is basically print to stdio and nothing more.
Also usb stdio is specifically enabled in the CMakefile so that too suggests it should be functional.

Eg
```
/* Uncomment this to wait forever for the USB connection if using stdio
 * over USB rather than via UART */
// #define PICO_STDIO_USB_CONNECT_WAIT_TIMEOUT_MS   -1
```

Aside: I am actually trying to get this to work right now and setting those values pre or post #include, pre main() isn't working for me. (although a sleep_ms(1000); just after stdio_init_all() does work.

lbt, 2022-07-04:
never mind - I completely misunderstood what was going on here. This needs to be set in the CMakeLists because it needs to build stdio_usb.c ... d'oh...

lbt, 2022-07-04:
I think the problem you have here is that you have told me, the reader, that usb stdio is going to work by defining:
```pico_enable_stdio_usb(test 1)```
and yet to actually get any output we need to know some fairly complex cmake syntax.
If you define a -1 value for PICO_STDIO_USB_CONNECT_WAIT_TIMEOUT_MS by default then uart users will get a hang :( 

I sugges the CMakeLists.txt should have something more like this (assuming I got it right this time !) : 

```
# uncomment the following lines to enable stdio over usb
# pico_enable_stdio_usb(test 1)
# target_compile_definitions(test PRIVATE
# PICO_STDIO_USB_CONNECT_WAIT_TIMEOUT_MS=10000
# )
```

The value of the timeout could be -1 but at least this way, confused uart users will see output eventually (and noob SDK users like me who have come straight in with a Pico W, will have a much easier time).

lurch, 2025-07-09:
Cloisng as duplicate of https://github.com/raspberrypi/pico-examples/issues/321

closed 2025-07-09 by lurch