raspberrypi/pico-feedback #260: Provide picow_blink binary in C SDK page

tbialas, 2022-07-14:
Dear Pi maintainers,

A minor issue I've run into as a new user of the Pico W is that the documentation for the C/C++ SDK 
in [documentation/asciidoc/microcontrollers/c_sdk/your_first_binary.adoc](https://github.com/raspberrypi/documentation/blob/4ffb8d3667cb814607dea0f76ad6d986dd0a04db/documentation/asciidoc/microcontrollers/c_sdk/your_first_binary.adoc)

The Blink binary will only work for the non-W variant of the Pico, as the LED on the PicoW is tied to the wireless chipset.
Would it be possible to also provide the binary for `picow_blink`, to allow users to check their board is working? I originally thought my board was defective after trying the example!

Thank you very much in advance,
Best regards,
Tomasz

aallan, 2022-07-19:
Done, see https://www.raspberrypi.com/documentation/microcontrollers/c_sdk.html#blink-an-led.

closed 2022-07-19 by aallan

lurch, 2022-07-19:
> see https://www.raspberrypi.com/documentation/microcontrollers/c_sdk.html#blink-an-led

Perhaps "You can see the code on Github" should also link to both the Pico and Pico W variants?

aallan, 2022-07-19:
Done f3ca7cf.