raspberrypi/pico-feedback #262: Request for OTA bootloader

estrellar, 2022-07-17:
OTA example for pico w.

aallan, 2022-07-17:
Writing an over-the-air bootloader for Pico W isn't really adding an example, it's a serious infrastructure project. We're a small team and it's very unlikely we'll get around to it ourselves. However, you should go look around [the forums](https://forums.raspberrypi.com/viewforum.php?f=143&sid=0fa11313605d43190fc3369b630cc07d), I know there is some community interest in writing one.

closed 2022-07-17 by aallan

lurch, 2022-07-17:
Coincidentally, see https://github.com/raspberrypi/pico-sdk/issues/928

usedbytes, 2022-07-17:
I spent my weekend porting over a bootloader I already wrote for regular Pico to use WiFi on Pico-W, see here:

https://github.com/usedbytes/picowota_blink (as referenced from #928)