raspberrypi/pico-feedback #265: Datasheet: USB device mode OUT handling

pigrew, 2022-07-22:
Some clarification in the the datasheet would be appreciated for the handling of USB OUT packets while in device mode.

For the buffer control register section `4.1.2.5.4`:

1. Are bits 13 and 29 (the PID / data toggle bits), ever set by the peripheral? Or is it only set by the CPU, and used by the peripheral to determine if a sequence error happened?

For the description of OUT transfers `4.1.2.6.3`:

1. If the endpoint is not enabled, does the device send a status token back to the host?
2. If a `DATA_SEQ_ERROR` happens, the USB spec says that the peripheral should ACK the packet and discard the transferred data. Does the peripheral discard the data? Or transfer it into the user buffer and set the PID in the control register? (I believe that TinyUSB has a bug if it is transferred, as it never checks the PID bits of the buffer control register after a transfer happens.)
3. What happens if `ENABLED && !(AVAILABLE && !FULL) && !STALL`? The peripheral should NAK the packet.

Perhaps it should also be made clear that the peripheral should always enter the data phase... but only store the data in the buffer if `(AVAILABLE && !FULL)`.

(This issue was prompted by discussion in https://github.com/hathach/tinyusb/issues/1572)

aallan, 2022-07-22:
Sorry, which datasheet are you referring to, there are lots?

pigrew, 2022-07-22:
RP2040 datasheet: https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf

lurch, 2022-07-22:
ping @liamfraser

mjan-jansson, 2022-07-23:
I believe point 3 above, regarding device OUT packets is related to the EP_ABORT and EP_ABORT_DONE endpoint control register. 

The documentation is ambiguous as to who (the cpu or the controller) sets the bit in the EP_ABORT_DONE register.  I would suggest updating the docs to make it explicit, and also to reference both EP_ABORT and EP_ABORT_DONE register in section 4.1.2.6.3 (if my assumptions are correct of course).

As a side note, to me it would have made more sense to call this register EP_BUSY,  EP_BLOCKED, EP_FORCE_NAK. Current name suggest that the endpoint is in an error state (aborted) and not simply ready to accept more data (busy).

pigrew, 2022-07-23:
My question was more related to what happens if the response token (i.e. the ACK) is corrupted while being sent to the host, and the host retries the packet, leading to a a sequence error.

This is mentioned in the USB 2.0 spec section 8.6.4:

>A lost or corrupted ACK handshake can lead to a temporary loss of synchronization between transmitter and receiver as shown in Figure 8-47.

>The receiver’s sequence bit and the data PID will not match, so the receiver knows that it has previously accepted this data. Consequently, it discards the incoming data packet and does not toggle its sequence bit.

The purpose of this issue is to confirm that TinyUSB is properly handling these sequence errors. Actual testing may be difficult (I may have to electrically inject faults), but hoped that first I could analyze the situation based on the documentation.

Updates to how aborting is handled are also appreciated (as an aside, TinyUSB does not support device-mode aborts.)