raspberrypi/pico-feedback #268: sleep_ms Stops the Core 1 Execution

BalajiDany, 2022-08-08:
I Tried Blink the LED inside core_1. Led is not switching off. and seems after sleep_ms core_1 is paused. Is this Known issue?

FYI.

```
#include "pico/stdlib.h"
#include "pico/multicore.h"

void core1_entry()
{
    const uint LED_PIN = PICO_DEFAULT_LED_PIN;
    gpio_init(LED_PIN);
    gpio_set_dir(LED_PIN, GPIO_OUT);

    while (1)
    {
        gpio_put(LED_PIN, 1);
        sleep_ms(250);
        gpio_put(LED_PIN, 0);
        sleep_ms(250);
    }
}

int main()
{
    multicore_launch_core1(core1_entry);
}
```

BalajiDany, 2022-08-08:
But #248 is Concluded as openocd Issue https://github.com/raspberrypi/pico-feedback/issues/248#issuecomment-1171463347

lurch, 2022-08-08:
I don't have time to investigate this myself, so ping @kilograham in case he can spot what's going on.

kilograham, 2022-08-08:
your problem is likely that you have returned from main and hit a breakpoint... if you were to do a while (true) {} in main you'll be fine

BalajiDany, 2022-08-08:
@kilograham Thanks for your support. It is working now!

closed 2022-08-08 by BalajiDany