raspberrypi/pico-feedback #269: getting started with pico :  update DOC

albmef, 2022-08-12:
Hi All !

getting-started-with-pico v1.8
ubuntu 20 LTS
Open On-Chip Debugger 0.11.0-g228ede4-dirty (2022-08-11-18:23)
pico-sdk  v1.4.0
VSCODE v1.70.1
cortex-debug v1.4.4

doc sections :
vscode + OOCD + GDB + picocom.

- build openocd : issue with git option "--no-single-branch"
the one liner below works
git clone https://github.com/raspberrypi/openocd.git --branch rp2040 --depth=1

- gdb-multiarch missing
add apt install gdb-multiarch

- useless option -s in   "openocd -f interface/picoprobe.cfg -f target/rp2040.cfg -s tcl"

launching openocd with the -s tcl option is useless because once openocd is installed with sudo make install,
the scripts are found in
/usr/local/bin/../share/openocd/scripts/target/rp2040.cfg
and
/usr/local/bin/../share/openocd/scripts/interface/picoprobe.cfg

- wrong VSCODE extension
microsoft cpp-tools does not exist in the repos
replace microsoft "cpp-tools" by "C/C++"
  

- cortex-debug : objdump not found when using gdb-multiarch 
sudo apt install binutils-arm-none-eabi

add
    "cortex-debug.objdumpPath": "/usr/bin/arm-none-eabi-objdump",
    "cortex-debug.gdbPath": "gdb-multiarch"
in settings.json

***settings.json***
{
    // These settings tweaks to the cmake plugin will ensure
    // that you debug using cortex-debug instead of trying to launch
    // a Pico binary on the host
    "cmake.statusbar.advanced": {
        "debug": {
            "visibility": "hidden"
        },
        "launch": {
            "visibility": "hidden"
        },
        "build": {
            "visibility": "hidden"
        },
        "buildTarget": {
            "visibility": "hidden"
        }
    },
    "cmake.buildBeforeRun": true,
    "C_Cpp.default.configurationProvider": "ms-vscode.cmake-tools"
    "cortex-debug.objdumpPath": "/usr/bin/arm-none-eabi-objdump",
    "cortex-debug.gdbPath": "gdb-multiarch"
*****

- tested launch.json for picoprobe
please find below two working debug conf.

*** launch.json***
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Pico Debug GDB ext",
            "type":"cortex-debug",
            "cwd": "${workspaceRoot}",
            "executable": "${command:cmake.launchTargetPath}",
            "request": "launch",
            "servertype": "external",
            // This may need to be arm-none-eabi-gdb depending on your system
            // Connect to an already running OpenOCD instance
            "gdbTarget": "127.0.0.1:3333",
            "svdFile": "${env:PICO_SDK_PATH}/src/rp2040/hardware_regs/rp2040.svd",
            "runToMain": true,
            // Work around for stopping at main on restart
            "postRestartCommands": [
                "break main",
                "continue"
            ]
        },
        {
            "name": "Pico Debug GDB int",
            "cwd": "${workspaceRoot}",
            "executable": "${command:cmake.launchTargetPath}",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            // This may need to be arm-none-eabi-gdb depending on your system
            "device": "RP2040",
            "configFiles": [
                "interface/picoprobe.cfg",
                "target/rp2040.cfg"
            ],
            "svdFile": "${env:PICO_SDK_PATH}/src/rp2040/hardware_regs/rp2040.svd",
            "runToMain": true,
            // Work around for stopping at main on restart
            "postRestartCommands": [
                "break main",
                "continue"
            ]
        }
    ]
}
**********

best regards

lurch, 2022-08-12:
I don't have time to look at this myself right now (and I'm not sure if anybody else does either), but I wonder if anything here is useful for @JamesH65 's [pico-project-generator](https://github.com/raspberrypi/pico-project-generator) ?

albmef, 2022-08-12:
Hi @lurch @JamesH65 

yes i created a very basic vscode project using pico-project-generator
i can reproduce the issue with missing objdump

here the big C code : 
#include <stdio.h>
#include "pico/stdlib.h"
int main()
{
    stdio_init_all();

    puts("Hello, world!");

    return 0;
}

below the debug console output :  

eading symbols from objdump-multiarch --syms -C -h -w /home/alban/Bureau/PICO/hello/build/hello.elf
Reading symbols from nm-multiarch --defined-only -S -l -C -p /home/alban/Bureau/PICO/hello/build/hello.elf
Launching GDB: gdb-multiarch -q --interpreter=mi2 /home/alban/Bureau/PICO/hello/build/hello.elf
    Set "showDevDebugOutput": true in your "launch.json" to see verbose GDB transactions here. Helpful to debug issues or report problems
Error: nm-multiarch failed! statics/global/functions may not be properly classified: Error: spawn nm-multiarch ENOENT
    Expecting `nm` next to `objdump`. If that is not the problem please report this.
Error: objdump failed! statics/globals/functions may not be properly classified: Error: spawn objdump-multiarch ENOENT    ENOENT means program not found. If that is not the issue, please report this problem.Launching gdb-server: openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/alban/Bureau/PICO/hello -f /home/alban/.vscode/extensions/marus25.cortex-debug-1.4.4/support/openocd-helpers.tcl -f interface/picoprobe.cfg -f target/rp2040.cfg
    Please check TERMINAL tab (gdb-server) for output from openocd
Finished reading symbols from objdump: Time: 63 ms
Finished reading symbols from nm: Time: 54 ms

it seems that cortex-debug expect nm-multiarch or objdump-multiarch to be installed
but AFAIK no debian package provide these files
that why i added the path to arm-none-eabi-objdump in the settings.json file

so if i add the two lines in the generated settings.json files
the file looks like this : 
{
  "cmake.configureOnOpen": false,
  "cmake.statusbar.advanced": {
    "debug" : {
      "visibility": "hidden"
              },    "launch" : {
      "visibility": "hidden"
               },
    "build" : {
      "visibility": "hidden"
               },
    "buildTarget" : {
      "visibility": "hidden"
               },
     },
    "cortex-debug.objdumpPath": "/usr/bin/arm-none-eabi-objdump",
    "cortex-debug.gdbPath": "gdb-multiarch"
}

and debug console does not show any warning anymore : 

Reading symbols from /usr/bin/arm-none-eabi-objdump --syms -C -h -w /home/alban/Bureau/PICO/hello/build/hello.elf
Reading symbols from /usr/bin/arm-none-eabi-nm --defined-only -S -l -C -p /home/alban/Bureau/PICO/hello/build/hello.elf
Launching GDB: gdb-multiarch -q --interpreter=mi2 /home/alban/Bureau/PICO/hello/build/hello.elf
    Set "showDevDebugOutput": true in your "launch.json" to see verbose GDB transactions here. Helpful to debug issues or report problems
Launching gdb-server: openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/alban/Bureau/PICO/hello -f /home/alban/.vscode/extensions/marus25.cortex-debug-1.4.4/support/openocd-helpers.tcl -f interface/picoprobe.cfg -f target/rp2040.cfg
    Please check TERMINAL tab (gdb-server) for output from openocd
Finished reading symbols from objdump: Time: 97 ms
Finished reading symbols from nm: Time: 137 ms

hope this helps
in the mean time if you find a better workaround feel free to share

Best regards

aallan, 2023-03-02:
Should be resolved with documentation updates since originally posted.

closed 2023-03-02 by aallan

hannah-m-moore, 2023-09-08:
> Should be resolved with documentation updates since originally posted.

I'm afraid not...I still get the same problem and I have tried to follow every tutorial as closely as possible.  Its positively doing my nut.  I know it has to be something inside vscode, or a vscode configuration file, somewhere because I can execute them all from the command line and they run...there are also a couple of extra flags being reported on the launch of gdb-multiarch which I cannot find defined anywhere so I know that they have to have been sourced from somewhere other than my paltry config files. 
 
Here is the error output minus the popup window:

Cortex-Debug: VSCode debugger extension version 1.12.0 git(d7a405a). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from /usr/bin/arm-none-eabi-objdump --syms -C -h -w /home/rabbit/workspace/RP2040/pico/pico-examples/build/hello_world/serial/hello_serial.elf
Reading symbols from /usr/bin/arm-none-eabi-nm --defined-only -S -l -C -p /home/rabbit/workspace/RP2040/pico/pico-examples/build/hello_world/serial/hello_serial.elf
Launching GDB: gdb-multiarch -q --interpreter=mi2
1-gdb-version
Launching gdb-server: /usr/local/bin -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/rabbit/workspace/RP2040/pico/pico-examples -s /usr/local/bin/tcl -c "adapter speed 5000" -f /home/rabbit/.vscode/extensions/marus25.cortex-debug-1.12.0/support/openocd-helpers.tcl -f /interface/cmsis-dap.cfg -f /target/rp2040.cfg
    Please check TERMINAL tab (gdb-server) for output from /usr/local/bin
Failed to launch OpenOCD GDB Server: Error: spawn /usr/local/bin EACCES

Here are my launch and settings files:

launch.json :
{
    "version": "0.2.0",
    "configurations": [
      {
        "name": "Pico Debug",
        "showDevDebugOutput": "raw",
        "device": "RP2040",
        "gdbPath": "gdb-multiarch",
        "cwd": "${workspaceRoot}",
        "executable": "${command:cmake.launchTargetPath}",
        "request": "launch",
        "type": "cortex-debug",
        "servertype": "openocd",
        "configFiles": [
          "/interface/cmsis-dap.cfg",
          "/target/rp2040.cfg",
        ],
        "searchDir": [
          "${workspaceRoot}",
          "${env:OPENOCD_PATH}/tcl",
        ],
        "openOCDPreConfigLaunchCommands": ["adapter speed 5000"],
        "svdFile": "${env:PICO_SDK_PATH}/src/rp2040/hardware_regs/rp2040.svd",
        "runToEntryPoint": "main",
        "postRestartCommands": [
          "break main",
          "continue",
        ]
      }
    ]
  }

Here is my settings.json:
{
    // These settings tweaks to the cmake plugin will ensure
    // that you debug using cortex-debug instead of trying to launch
    // a Pico binary on the host
    "cmake.statusbar.advanced": {
        "debug": {
            "visibility": "hidden"
        },
        "launch": {
            "visibility": "hidden"
        },
        "build": {
            "visibility": "default"
        },
        "buildTarget": {
            "visibility": "hidden"
        }
    },
    "cmake.buildBeforeRun": true,
[... 93 lines cut ...]
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from hello_serial.elf...done.
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
time_reached (t=...)
    at /home/rabbit/workspace/RP2040/pico/pico-sdk/src/rp2_common/hardware_timer/include/hardware/timer.h:116
116         uint32_t hi_target = (uint32_t)(target >> 32u);
(gdb) monitor reset init
[rp2040.core0] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
[rp2040.core1] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
(gdb) continue
Continuing.

and of course I can see "Hello World" in my minicom...
I cannot seem to find where I've gone wrong anywhere...other than choosing to use another microsoft product with their typically rubbish documentation.

Can someone please point out where I'm doing something stupid...I've been at this for over two weeks now.

Cheers, 
Sasha