raspberrypi/pico-feedback #272: Documentation (rp2040 data sheet): no "List of Registers" in Table-of-Contents entry for SIO

WestfW, 2022-08-29:


lurch, 2022-08-30:
I agree that it's unfortunate that the "List of Registers" for SIO doesn't show up in the ToC. However getting the right "depth" of headers displayed in the ToC is a bit of a balancing-act; as you may have guessed the ToC is entirely auto-generated. If we increased the ToC-depth so that the SIO's List-of-Registers _did_ show up in the ToC, this would also include **lots** of other headers into the ToC, and this would make the ToC so long as to become unwieldy.

closed 2022-08-30 by lurch

WestfW, 2022-08-30:
> as you may have guessed the ToC is entirely auto-generated.

Yeah, I figured.  Are you sure that there isn't a way to explicitly add a ToC entry, bypassing the automation?  It seems like it would be a pretty common requirement (though I am not at all in touch with modern Document Preparation Tools.)

lurch, 2022-08-31:
> Are you sure that there isn't a way to explicitly add a ToC entry, bypassing the automation?

Yes. https://docs.asciidoctor.org/asciidoc/latest/toc/

We spent some time juggling the ToC level before the release of the RP2040 and its documentation, and settled on the current value as the best compromise.

Although I've just done a bit of additional searching and found https://github.com/asciidoctor/asciidoctor-pdf/issues/2097 so we'll discuss internally if this is a feature we want to implement. (It's a feature only available in AsciiDoctorPDF 2.x, and we're also currently migrating our toolchain from AsciiDoctorPDF 1.x to AsciiDoctorPDF 2.x)