raspberrypi/pico-feedback #273: Getting Started Guide - 2.2. Install the Toolchain - Needs note on Msys install of mingw...cmake

drankinatty, 2022-09-13:
After going through the SDK getting started for both Linux and Windows installs, Section "2.2. Install the Toolchain" specifies the install of cmake. That's fine on Linux, but under Windows, installing the normal cmake (which is available for Msys2) will result in build failures with:

``` none
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
System is unknown to cmake, create:
Platform/MINGW64_NT-10.0-19044 to use this system, please post your config file on discourse.cmake.org so it can be adde d to cmake
-- Detecting C compiler ABI info
System is unknown to cmake, create:
Platform/MINGW64_NT-10.0-19044 to use this system, please post your config file on discourse.cmake.org so it can be adde d to cmake
...
```

and then a slew of linker errors:

```none
[100%] Linking CXX executable elf2uf2
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/elf2uf2 .dir/main.cpp.obj:main.cpp:(.xdata+0x5c): undefined reference to `__gxx_personality_seh0'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/elf2uf2 .dir/main.cpp.obj:main.cpp:(.xdata+0xbc): undefined reference to `__gxx_personality_seh0'
...
```

The problem is there is also a specific `mingw-w64-x86_64-cmake` which must be installed instead of the generic cmake. It would really help Windows users if this was noted in Section "2.2. Install the Toolchain" of the getting started guide.

lurch, 2022-09-13:
At the very start of Chapter 2 it says "They also assume you are using Raspberry Pi OS running on a Raspberry Pi 4,
or an equivalent Debian-based Linux distribution running on another platform. Alternative instructions for those
using Microsoft Windows (see Section 9.2) or Apple macOS (see Section 9.1) are also provided.". And the toolchain-installation instructions for Windows are specifically in section 9.2.1

If you want to build pico-sdk and the examples on Windows using Msys2, rather than following the instructions we provide in the getting-started guide, then that's an unsupported configuration and you'll need to ask for help at https://forums.raspberrypi.com/viewforum.php?f=143

closed 2022-09-13 by lurch

drankinatty, 2022-09-14:
Thanks for taking a look. Just trying to improve how a bulk of people will come at this. It works really well. As explained in the issue, I don't need help, I solved the issue after a couple of searches knowing what I was looking for. However, someone excited to get going with the Pico without 35 years of computing experience would benefit from a note -- even a footnote pointing to section 9.2.1. 

Any plans on adding to the SDK docs to expand on the hardware function descriptions? Some a barely more than a restatement of the declarations with a bullet point for each of the parameters and return (often describing the return on success only, without explaining what is returned on failure). Food for thought. Not a criticism. I know manhours don't materialize out of thin-air. Keep up the great work.

lurch, 2022-09-14:
> Just trying to improve how a bulk of people will come at this.

We expect that the bulk of people will come at this by reading the chapter specifically dedicated to installing the toolchain on Windows :wink: It seems unlikely that somebody "without 35 years of computing experience" would even be using Msys2 in the first place?

> Any plans on adding to the SDK docs to expand on the hardware function descriptions? Some a barely more than a restatement of the declarations with a bullet point for each of the parameters and return

The SDK function descriptions in the [C SDK PDF](https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf) are automatically generated from the [doxygen](https://doxygen.nl/)-comments in the source-code, and _should_ contain exactly the same information as https://raspberrypi.github.io/pico-sdk-doxygen/index.html - if you spot any discrepancies please let us know. If there are any functions in particular that you feel need a more detailed explanation, then please open an issue at https://github.com/raspberrypi/pico-sdk/issues (and see also https://github.com/raspberrypi/pico-sdk/issues?q=is%3Aopen+is%3Aissue+label%3Adocumentation and https://github.com/raspberrypi/pico-feedback/issues?q=is%3Aopen+is%3Aissue+label%3Adocumentation )

> I know manhours don't materialize out of thin-air. Keep up the great work.

Thanks. We're only a relatively small engineering team, and we're always very busy :sweat_smile: