raspberrypi/pico-feedback #274: [Documentation] libusb-compat needs to be added to the Mac OpenOCD build requirements

ecurtz, 2022-09-16:
OpenOCD apparently uses both libusb and libusb-1 so it requires the shims in libusb-compat to build on the Mac (at least on M1).

aallan, 2022-09-16:
I built openocd on an Intel Mac the day before yesterday, and didn't need to do this, so could be an M[1,2] specific problem?

ecurtz, 2022-09-16:
Very possible that it's M[1,2] only. I was getting link errors on the bulk_endpoint functions from "original" libusb and looking at the generated Makefile it had both LIBUSB0_CFLAGS / LIBUSB0_LIBS
and LIBUSB1_CFLAGS / LIBUSB1_LIBS with the LIBUSB0 ones pointing to an old x86 version (built locally by me). Adding libusb-compat to the ARM64 brew installs and reconfiguring/building fixed it.

aallan, 2022-10-11:
Just built for an M1 based Mac and I didn't need to do this. I followed the instructions as written, and they worked for me on a clean machine.

closed 2022-10-11 by aallan