raspberrypi/pico-feedback #275: RP2040 Datasheet Pg 417 Figure 59 UART Block Diagram (copy/paste error)

drankinatty, 2022-09-20:
In the RP2040 Datasheet Page 417 in Figure 59 in the DMA Interface (bottom-left of figure) the first **UARTTXDMABREQ** should be the Receive Burst Request **UARTRXDMABREQ**. If you look at the figure there are two **UARTTXDMABREQ** shown leaving the DMA Interface block and no **UARTRXDMABREQ**.

The bottom **UARTTXDMABREQ** is correct, but the **UARTTXDMABREQ** (2nd up from the bottom) should be **UARTRXDMABREQ**. (it just looks like something got copied and was intended to be changed, but then life intervened and was missed)

(this is low-priority, but as I find issues, I'll go ahead and report them so they can be fixed as time permits -- or if I learn how the documentation works I'll make a pull-request)

lurch, 2022-09-20:
> but as I find issues, I'll go ahead and report them so they can be fixed as time permits

Thanks! Issues with the RP2040 / Pico documentation should be filed at https://github.com/raspberrypi/pico-feedback/issues

> or if I learn how the documentation works I'll make a pull-request

The source for https://www.raspberrypi.com/documentation is available at https://github.com/raspberrypi/documentation/tree/develop/documentation/asciidoc (and PRs are indeed welcome), whereas the source for the RP2040 & Pico PDFs isn't publicly available (but please feel free to file issues, and we'll fix them as time permits).

lurch, 2025-07-09:
To clarify for the benefit of future readers: this issue is pointing out that this part of Figure 59 in the RP2040 datasheet

[image]

ought to say **UARTRXDMABREQ**, as it does in Arm's PL011 datasheet:  (literally a one-character change!)

[image]

lurch, 2025-07-10:
And (as expected) the same minor image-typo is also present in the RP2350 datasheet

[image]