raspberrypi/pico-feedback #276: Getting Started with Pico: pkg-config required to build OpenOCD

crosland, 2022-09-22:
Appendix A of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf the instructions for building on Linux should include a requirement for pkg-config to be installed, i.e. add to the 'sudo apt install ...' step.

Ubuntu 20.04 on WSL2 does not seem to include pkg-config and the configure step will show errors and bootstrap will fail

```
~/pico/openocd$ ./bootstrap
+ aclocal --warnings=all
+ libtoolize --automake --copy
+ autoconf --warnings=all
configure.ac:13: error: possibly undefined macro: AC_MSG_WARN
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:212: error: possibly undefined macro: AC_DEFINE
configure.ac:353: error: possibly undefined macro: AC_MSG_ERROR
configure.ac:652: error: possibly undefined macro: AC_MSG_NOTICE
```

after
`sudo apt install pkg-config`
rerun configure and bootstrap ind there are no errors.

lurch, 2022-09-22:
I just checked, and Appendix A _does_ already say "Ubuntu and Debian users might additionally need to also install pkg-config."

closed 2022-09-22 by lurch

crosland, 2022-09-22:
Oops, sorry. It would be better if were done as a footnote like "1. if you ar ebuilding ..." to draw attention to the not ebeing on the next page.

lurch, 2022-09-22:
We _could_ make it a footnote, but with the way that the PDFs get built there's no guarantee that the footnote wouldn't end up on the next page too.

crosland, 2022-09-22:
> From "Andrew Scheller" ***@***.***>
> To "raspberrypi/pico-feedback" ***@***.***>
> Cc "Andrew Crosland" ***@***.***>; "Author"
> Date 22/09/2022 12:33:19
> Subject Re: [raspberrypi/pico-feedback] Getting Started with Pico:
> pkg-config required to build OpenOCD (Issue #276)

But at least there would be a link to it.