raspberrypi/pico-feedback #277: Simple question on PIO CTRL register's SM_RESTART

henrygab, 2022-09-30:
When starting (or restarting) a PIO state machine, is there an **_architecturally guaranteed_** starting value for the `X` and `Y` registers?

Alternatively, are those registers architecturally guaranteed to retain the last value they had from any prior execution?

### Background on my question...

I was working on the presumption that the `X` and `Y` registers are initialized to zero by hardware when starting / restarting a PIO state machine... but then I noticed that these registers were not listed in the 3.7 "List of Registers", `CTRL` register, in the documentation on what's reset when the `SM_RESTART` bit(s) is(are) written.  The documentation states:

> Specifically, the following are cleared: input and output
> shift counters; the contents of the input shift register; the
> delay counter; the waiting-on-IRQ state; any stalled
> instruction written to SMx_INSTR or run by OUT/MOV
> EXEC; any pin write left asserted due to OUT_STICKY.

The absence of the `X` and `Y` registers being cleared to zero is noticeably absent.

I also failed to notice any mention of their initial state in 3.2 "programmer's model".

Thus, I'm looking for explicit official writing indicating what I *should* expect, so I can adjust my expectations (and PIO assembly) accordingly.

Thanks!

Wren6991, 2022-09-30:
Note also SM_RESTART does not affect the contents of the OSR, since this is expected to be pulled into before the first OUT instruction.

However it does clear the ISR contents (as mentioned in the register doc string) because part of the ISR's initial value will be written to the FIFO on the first PUSH.

henrygab, 2022-09-30:
Hi @Wren6991,

Thanks for the clear answers!  I love learning!

I think there may be more than one PIO assembly program out there that presumed the registers both started at zero.  I know I have to now go and update some programs.  🙂 

FWIW, I think it's the right decision at the hardware level ... **_not_** zero'ing the registers seems to allow for swapping the program for a SM, while retaining the prior program's result in `X` and `Y`, which might open interesting options.

**_Thank you_** for offering to update SM_RESTART to also list what **_doesn't_** change.

As a concise way to convey this information, would it make sense to add (maybe in the "Programmer's Model") a table listing what the architectural guarantees for state are, when starting a program?  I'd like to avoid others being surprised by having assumptions on starting state, especially as it's extremely difficult to track down bugs which are rooted in mistaken assumptions.   (Over-specification is better than under-specification.)

Item | hard reset | SM_RESTART | ...
|-----|-----|-----|-----
| `X` register | zero | last value
| `Y` register | zero | last value
| `ISR` register | zero | zero | 
| `ISR` counter | zero | zero |
| `OSR` register | zero | (?!) last value |
| `OSR` counter | zero | (?!) zero |
| Delay counter | zero | zero | 
| Waiting-on-IRQ state | false | false | 
| SMx_INSTR / OUT EXEC / MOV EXEC instruction | none | none |
| OUT_STICKY state |  ... | ... 
| ... etc.

Thanks!

mjcross, 2023-04-14:
Glad I read this thread, I was just about to commit some code that assumed `pio_sm_init()` zeroes the scratch registers.

For what it's worth I'm now initialising the X register with the following (before enabling the state machine):

```
    // initialise X register to zero
    pio_sm_exec_wait_blocking (pio, sm, pio_encode_mov (pio_x, pio_null));
```

kilograham, 2025-07-08:
Closing as the documentation was updated

closed 2025-07-08 by kilograham