raspberrypi/pico-feedback #280: No timing information for PIO (RP2040 datasheet)

arg08, 2022-10-10:
The RP2040 datasheet does not provide any timing information for the PIO.

In particular, the PIO provides the INPUT_SYNC_BYPASS feature, which requires input data to meet setup/hold times to avoid metastability, but these times are not specified (and note also that INPUT_SYNC_BYPASS is used in the pico-examples SPI implementation).

Even if INPUT_SYNC_BYPASS is not used, this information is still important for any synchronous application of PIO that wants to achieve maximum performance - the input synchronisers trade metastability for uncertainty as to when (ie. which cycle) the input will become valid and so can be sampled reliably by the PIO program.

Presumably the required setup/hold are relative to CLK_SYS (which is not externally visible).  Probably the most common case for synchronous PIO timing is where the clock is generated as a PIO output and the data is generated by external hardware in response to that clock (the SPI example falls into this category).

This could be satisfied by documenting required setup/hold times defined for PIO  inputs relative to CLK_SYS, and output delays from CLK_SYS to PIO outputs changing with an out/set instruction.

Ideally, delays would also be specified so that systems using CLK_GPOUTn or GPCLKn can relate timing of those clocks to CLK_SYS.

Wren6991, 2022-11-01:
Sorry for the delay. I'll summarise some information here, and once you are happy you have enough details, we can add this to the datasheet.

# Digital behaviour

There is a register stage between each state machine and the pin mapping logic, and a register stage between the input mapping logic and each state machine. Assuming zero propagation delay, there would be a two-cycle delay in a state machine observing its own outputs, if synchronisers are bypassed, or a four-cycle delay, if synchronisers are engaged. For example if you had a program like this:

```c
    // Fill instruction memory with IN PINS 1, and a sideset of either 1 or 0.
    for (int i = 0; i < 32; ++i)
        pio->instr_mem[i] = i < 16 ? 0x5001 : 0x4001;
```

This would put the value `0xffff0000u << 2` in the ISR (synchronisers bypassed) or `0xffff0000u << 4` (synchronisers enabled).

# Setup and Hold

The flops themselves have a typical setup time of 10.6 ps and hold time of 2.2 ps. More importantly, the IO delays between flops and pads must be taken into account.

# IO delays

These delays include PIO's input/output mapping logic, IO muxing, and the actual pad delays into a nominal load of a few pF. Min/max is over the extremes of process variation, voltage (1.1 V +- 10%) and temperature (-40 C to 125 C).

These delays assume an IOVDD of 1.8 V, with PADS_VSEL set. At IOVDD = 3.3 V, the delay is significantly lower, and the range is smaller.

Min/max output delays, from clk_sys arriving at any flop in PIO to the data being valid at a particular GPIO pad:

Pad output | Min delay (ns) | Max delay (ns)
-----------|----------------|---------------
GPIO0      | 2.27 | 7.10
GPIO1      | 2.31 | 7.07
GPIO2      | 2.33 | 7.08
GPIO3      | 2.24 | 7.00
GPIO4      | 2.30 | 7.07
GPIO5      | 2.34 | 7.10
GPIO6      | 2.32 | 7.10
GPIO7      | 2.39 | 7.09
GPIO8      | 2.34 | 7.09
GPIO9      | 2.38 | 7.08
GPIO10     | 2.33 | 7.07
GPIO11     | 2.36 | 7.08
GPIO12     | 2.35 | 7.04
GPIO13     | 2.31 | 7.08
GPIO14     | 2.38 | 7.06
GPIO15     | 2.33 | 7.05
GPIO16     | 2.34 | 7.09
GPIO17     | 2.37 | 7.09
GPIO18     | 2.37 | 7.04
GPIO19     | 2.27 | 7.10
GPIO20     | 2.38 | 7.09
GPIO21     | 2.05 | 7.10
GPIO22     | 2.34 | 7.07
GPIO23     | 2.16 | 7.05
GPIO24     | 2.12 | 7.06
GPIO25     | 2.26 | 7.10
GPIO26     | 2.32 | 7.09
GPIO27     | 2.26 | 7.08
GPIO28     | 2.34 | 7.09
GPIO29     | 2.30 | 7.07

Min/max input delays, from pad input to the input synchroniser:

Pad output | Min delay (ns) | Max delay (ns)
-----------|----------------|---------------
GPIO1      | 1.89 | 5.22
GPIO2      | 1.84 | 5.25
GPIO3      | 1.83 | 5.24
GPIO4      | 1.90 | 5.17
GPIO5      | 1.90 | 5.14
GPIO6      | 1.91 | 5.19
GPIO7      | 1.91 | 5.14
GPIO8      | 1.95 | 5.14
GPIO9      | 1.96 | 5.12
GPIO10     | 1.95 | 5.11
GPIO11     | 1.92 | 5.16
GPIO12     | 1.92 | 5.15
GPIO13     | 1.94 | 5.16
GPIO14     | 1.90 | 5.18
GPIO15     | 1.92 | 5.15
GPIO16     | 1.95 | 5.13
GPIO17     | 1.95 | 5.12
GPIO18     | 1.95 | 5.10
GPIO19     | 1.95 | 5.12
GPIO21     | 2.07 | 4.98
GPIO23     | 1.98 | 5.06
GPIO24     | 1.97 | 5.07
GPIO25     | 1.97 | 5.08
GPIO26     | 1.96 | 5.12
GPIO27     | 1.94 | 5.13
GPIO28     | 1.95 | 5.13
GPIO29     | 1.99 | 5.10

Min/max input delays over all corners, from pad input to state machine IN data flops (synchronisers bypassed):

Pad input | Min delay (ns) | Max delay (ns)
-----------|----------------|---------------
GPIO1      | 2.22 | 5.45
GPIO2      | 2.25 | 5.49
GPIO3      | 2.23 | 5.18
GPIO4      | 2.24 | 5.41
GPIO5      | 2.30 | 5.65
GPIO6      | 2.25 | 5.48
GPIO7      | 2.26 | 5.50
GPIO8      | 2.30 | 5.51
GPIO9      | 2.25 | 5.68
GPIO10     | 2.34 | 5.71
GPIO11     | 2.28 | 5.47
GPIO12     | 2.29 | 5.40
GPIO13     | 2.25 | 5.47
GPIO14     | 2.24 | 5.41
GPIO15     | 2.23 | 5.47
GPIO16     | 2.30 | 5.42
GPIO17     | 2.28 | 5.44
GPIO18     | 2.28 | 5.34
GPIO19     | 2.30 | 5.50
GPIO21     | 2.16 | 5.79
GPIO23     | 2.33 | 5.53
GPIO24     | 2.28 | 5.60
GPIO25     | 2.29 | 5.53
GPIO26     | 2.28 | 5.38
GPIO27     | 2.27 | 5.39
GPIO28     | 2.24 | 5.28
GPIO29     | 2.33 | 5.47

# Effects of IOVDD

All of the IO delays given above are with IOVDD = 1V8. Increasing IOVDD to 3V3 reduces the pad delays quite significantly, and the pad delay is a large fraction of the delays reported above. Here's a summary of best/worst pad delays at 1V8/3V3:

Path type | IOVDD | Min delay (ns) | Max delay (ns)
----------|-------|----------------|--------------
Output    | 1V8   | 1.54           | 3.65
Output    | 3V3   | 1.11           | 2.14
Input     | 1V8   | 0.63           | 1.06
Input     | 3V3   | 0.47           | 0.76

Changing IOVDD does not affect any logic in the core domain, so these differences can be added to the IO delay tables above to estimate the IO delay ranges at IOVDD = 3V3.

Path group    | IOVDD | Min delay (ns) | Max delay (ns)
--------------|-------|----------------|---------------
Output        | 1V8   | 2.12           | 7.10
Output        | 3V3   | 1.69           | 5.59
Input to sync | 1V8   | 1.83           | 5.25
Input to sync | 3V3   | 1.67           | 4.95
Input to SM   | 1V8   | 2.16           | 5.79
Input to SM   | 3V3   | 2.00           | 5.49

arg08, 2022-11-01:
This is excellent info, thank you!

The two-cycle delay on output probably wants to be flagged in the PIO chapter - there's text in 3.5.6.3 telling you about the 2 cycle delay on the input synchronizer so I was aware of that, but the pipeline delay on the outputs was news to me (though makes perfect sense now you mention it).  Maybe a note in 3.5.6.1?

The one thing that's ambiguous in what you've said here is the "2 clocks" of output delay - I'm assuming that this is clk_sys free running (ie. not affected by the clock gating that applies to the rest of the PIO logic when a clock divider is in use in a given SM).  Probably worth an explicit sentence to that effect.

This is now enough info for me to check whether my application is going to work - I'm currently planning to generate the clock out of the PIO itself (toggling on alternate instructions with side-set), so the data out timing is fixed relative to my generated clock and you've now given me the limits on skew between the various pins; input was always going to be more tricky but again I can now calculate the range of my output clock edges relative to CLK_SYS, add the delays through the external hardware and the input delays you've given me and check that I'm hitting the window.  In my application I will probably leave the input synchronisers enabled as I can live with the extra delay (but care deeply about the data arriving in a particular clock cycle!).

In an ideal world, we'd also have data to allow the other ways of implementing this to be analysed.  I could build my application as it is now, but instead of toggling my clock output from the PIO I could use one of the dedicated clock outputs (clk_gpout0) with clk_sys running from the PLL direct and clk_gpout0 being PLL/2.  Or an external clock could be fed to clk_gpin0, and used as clk_sys (though less useful as now clk_sys is limited to 50MHz).   These scenarios would need data on the delays through the various clock units - though they are more obscure cases and likely to be less performant than the pure PIO ones, so it would be quite reasonable if you can't be bothered to document this.

While we are on the topic of documenting previously-unknown delays through the PIO,  one other piece of timing is the IRQ between SMs.  The datasheet proudly says "This allows cycle-accurate synchronisation between state machines." (3.2.7) but doesn't specify the amount of delay.  I'm guessing that an IRQ instruction executes in an SM on one cycle, the WAIT instruction on the 2nd SM then completes its execution in the next cycle, and then the next useful instruction in the target SM executes in the cycle after that - but there could easily be extra delays I'm not aware of, and it would be nice if the documentation told us.

Wren6991, 2022-11-02:
> The one thing that's ambiguous in what you've said here is the "2 clocks" of output delay - I'm assuming that this is clk_sys free running (ie. not affected by the clock gating that applies to the [...]

To be clear, it's one cycle on output and one cycle on input. The round trip delay is two cycles (plus propagation delay, plus pad delay). And yes, this is clk_sys. The clock divider only affects the execution speed of the SM. Good point, I'll make sure to note that clearly.

> These scenarios would need data on the delays through the various clock units - though they are more obscure cases and likely to be less performant than the pure PIO ones, so it would be quite reasonable if you can't be bothered to document this.

This is not so simple to document because a clock edge reaches different internal flops at different times. The path from the system PLL to a PIO flop is itself around 8 ns. What exactly would you find useful? The delay difference between clk_sys reaching PIO's output registers, and clk_sys propagating to a gpout pad? The skew between PIO outputs reaching a pad and the corresponding clk_sys edge reaching a gpout pad?

> While we are on the topic of documenting previously-unknown delays through the PIO, one other piece of timing is the IRQ between SMs. The datasheet proudly says "This allows cycle-accurate synchronisation between state machines." (3.2.7) but doesn't specify the amount of delay.

When an SM executes an IRQ instruction, that IRQ flag is visible to all SMs on the next system clock cycle. A WAIT instruction will stall -- i.e., fail to execute, and attempt to execute again on the next clock enable for that SM -- on cycles where the IRQ is not yet set. So, assume all SMs have a clock divisor of 1, then assume that some SM A is waiting for an IRQ from SM B, and SM B executes an IRQ instruction. Cycle by cycle:

1. A is stalled on a WAIT instruction. B executes an IRQ instruction.
2. The IRQ flag is now seen to be high, so A's WAIT instruction executes successfully. B is executing its next instruction.
3. A is executing the instruction after the WAIT. B is executing its third instruction.

Similarly, when a WAIT IRQ instruction clears an IRQ, this is observable on the next system clock cycle.

1. A is stalled on a WAIT instruction. B executes an IRQ WAIT instruction
2. The IRQ flag is seen high, so the WAIT completes, and clears the IRQ. B is in the IRQ wait state, as the IRQ is not yet seen to be cleared again
3. A is executing the instruction after the WAIT. The IRQ flag is seen low again, so B's IRQ wait state ends this cycle.
4. A is executing its third instruction. B is executing the instruction after the IRQ WAIT.

Something to be careful of is that WAIT only checks the IRQs on cycles where the SM is clock-enabled, so if you have multiple SMs waiting for the same IRQ, they need to have the same clock divisor and their clock dividers need to be harmonised with CTRL_CLKDIV_RESTART

arg08, 2022-11-02:
> This is not so simple to document because a clock edge reaches different internal flops at different times. The path from the system PLL to a PIO flop is itself around 8 ns. What exactly would you [...]

Yes, as I say you've probably already covered at least 90% of the use cases that matter.  If you are willing to do more, it's a question of whether you cherry-pick the useful cases, or exhaustively document all the elements so that people can add up the bits to get what they need to know.

So for example there's the scenario of using gp_clk0 as an input clock directly as clk_sys - so you want to know the delay from the pin to clk_sys (and can then apply all the numbers given earlier to relate the PIO outputs to that input clock).  This scenario on the one hand is simple enough to make reasonable calculations - the info needed is just a single max/min delay spec, but on the other hand is relatively less useful as you are restricted to 50MHz and to one PIO cycle per input clock.

The scenario with the external clock feeding clk_sys via the PLL (has to be XIN now, not gp_clk0 (*)) is functionally more useful, but that delay from XIN edge to clk_sys edge via the PLL  probably harder to specify, and making use of this requires the PIO listening to that input clock to determine the phase; whether or not that actually works could then be calculated with the PIO input timings already given, but you might well argue that this is so esoteric as to not be worth supporting (and the stack of multiple delay tolerances may mean it usually doesn't work anyhow).

Then there's the output scenarios - PIO running from clk_sys, clk_gpoutN configured with the same source.  So what you want to know is the delay from clk_sys (as seen by the PIO) to the clk_gpoutN output pin, but that's actually composed of the delay from the source through the gpoutN clock generator to the pin, minus the delay through the clk_sys clock generator to the PIO.  If this number has a reasonably small range then it's potentially quite useful - saves using PIO resources to generate the output clock.  But if the sum of tolerances is so big that it's most of a cycle of a reasonably fast clock then you end up not really knowing anything about the relative phase of the PIO outputs and the clock output.

> When an SM executes an IRQ instruction, that IRQ flag is visible to all SMs on the next system clock cycle. A WAIT instruction will stall -- i.e., fail to execute, and attempt to execute again on the next clock enable for that SM -- on cycles where the IRQ is not yet set. 

Thanks for this  It works much as I had guessed, but it's good to have the details confirmed.

----

(*) BTW, I find this constraint that the PLL can only run from XIN rather limiting, mainly because the bootrom requires XIN as 12MHz so on the one hand you can't use XIN as an arbitrary clock due to the bootrom but you can't use one of the other pins to drive the PLL.  For example, I have a design where there's a TCXO running some radios and we'd like the RP2040 to be synchronous with that; we wouldn't mind the extra cost of a cheap 12MHz xtal to allow the bootrom to be used, but we can't have that without giving up the option to clock the CPU/PIO with a multiple of our accurate clock.   Nothing to be done about it, but thought I'd mention it as a piece of feedback for future devices.

arg08, 2023-06-04:
By way of follow-up, my concerns are satisfied and I have gone on to build a satisfactory product using them.  If you just added the tables out of this answer somewhere in chapter 5 of the datasheet I would be happy.

However, the issue has come up several times on the forums, most recently here:
https://forums.raspberrypi.com/viewtopic.php?p=2111449#p2111449

and it's evident that many users haven't heard of/don't understand metastability, and that the existing warnings are too vague to stop people doing bad things.

In particular, the note in 2.21.2 is not helpful:
"If you’re feeling brave, you can bypass to save two cycles of input latency."
Many Pico users are brave; not all of them know what they are doing.

I would suggest something like:
"If you are certain that the input data is guaranteed to meet the setup and hold times relative to CLK_SYS, then you can bypass to save two cycles of input latency".
(actually, this one is talking about separate synchronizers to SIO registers rather than PIO, so unless you are going to document those as well, this particular note should probably change to "you should not use this as it's not possible to ensure that the timing of the external data applied to the pin avoids metastability".

bruelltuete, 2023-07-03:
Having this info in the datasheet would be super helpful!
A worked clock-tick by clock-tick example like the PIO wait you have in a previous comment would be great.
If possible please also include a timing diagram like Petteri's, to illustrate where the various delays come in.
thanks!

aallan, 2023-07-03:
@Wren6991 Has this ended up on `develop` branch for the RP2040 Datasheet? Or is it not yet in there?

tybasuleman, 2024-02-08:
Can anyone help me out. I am trying to sync my 2 state machines with a clock signal so when clock is ON, State Machine one sends a signal and when it´s OFF, the other one sends a signal. Signal here is PDM bits. The problem I think I understand is that every instruction is carried out for one complete clock cycle, not half of it as per my requirement and I cannot say out pins, 0.5 or something per say. It has to be between 1 and 32.
So I am unable to turn the state machine off mid way. We cannot use two wait instructions in one pio file as well otherwise that would have done the job. If I am approaching the problem wrong then please correct me.
Thanks

lurch, 2024-02-08:
@tybasuleman See #375 :wink:

lurch, 2024-02-08:
I meant if you've already got one issue open, there's no need to go commenting the same thing on other issues too. (that's what I was trying to imply with the wink, apologies if it was too obscure)

tybasuleman, 2024-02-08:
Well, no one is replying and I really need to get this done. You are right, sorry.

aallan, 2024-02-08:
@nathan-contino Can you coordinate with @Wren6991 to get this into the RP2040 Datasheet? Thanks.

closed 2024-04-02 by nathan-contino