raspberrypi/pico-feedback #284: DMA INTR marked as RO but description suggests WC

daveythacher, 2022-10-18:
I was trying to use DMA interrupts to determine the status of a transfer. The interrupt was not driving the NVIC, so I knew I needed to look at the raw interrupts. However I was uncertain how to clear the interrupt, which in hindsight is probably not required. The SDK marks the INTR register as read only and will complain. However the datasheet suggests that this is valid usage despite also marking it as read only.

My guess is that clearing the INTR is handled by the module. Changing the state of the channel will change INTR. INTS is a register which changes value on the rising edge of INTR if enabled by INTE. Therefore INTS needs to be cleared so that it can report INTR to NVIC and so the CPU can be released from the NVIC.

Here is the what is written in the datasheet: (Table 137)
> Raw interrupt status for DMA Channels 0..15. Bit n corresponds to channel n.
Ignores any masking or forcing. Channel interrupts can be cleared by writing a
bit mask to INTR, INTS0 or INTS1.

lurch, 2022-10-18:
To clarify: I _think_ you're saying that it's inconsistent for this
[image]

to say "Channel interrupts can be cleared by writing a bit mask to INTR" whilst also labelling INTR itself as RO (read-only) ?

daveythacher, 2022-10-29:
Yes

daveythacher, 2022-10-29:
Yes, that is somewhat confusing. 

As the interrupt is automatically cleared in INTR. Writing to INTR makes no sense. Writing to INTSx is only required if enabled in INTEx.

Therefore the statement should be:
> Channel interrupts can be cleared by writing a bit mask to INTS0 or INTS1.

daveythacher, 2022-10-29:
Assuming I am correct.

Wren6991, 2022-11-02:
INTR, INTS0 and INTS1 are different views of the same underlying 12-bit interrupt register. INTR displays the raw 12-bit value, INTS0 masks the value with INTE0, and INTS1 masks the value with INTE1. The underlying interrupt flags (of which, again, there is a single flag per channel) are *set* by hardware, but must be *cleared* by software, because the DMA has no way of knowing when the processor has entered an interrupt handler.

Channel interrupts can be cleared by writing a bit mask to any of INTR, INTS0 and INTS1 -- the effect is the same.  The channel interrupt must be cleared, by any one of these mechanisms, before exiting the interrupt handler (customarily, acknowledging the interrupt is the first thing a handler does), otherwise you will immediately find yourself back in the handler again.

The fact it is listed as RO is a mistake, and I'll fix it.

Wren6991, 2022-11-02:
Having a look at the register description source, this was already fixed some time ago but the headers and datasheet have not yet been reexported. This will be fixed automatically in the next docs release.

daveythacher, 2022-11-04:
> INTR, INTS0 and INTS1 are different views of the same underlying 12-bit interrupt register. INTR displays the raw 12-bit value, INTS0 masks the value with INTE0, and INTS1 masks the value with [...]

This makes sense. One possible point of confusion is this design can glitch. To prevent this you would need a SR latch on INTSx. However I would recommend against this software design. 

Knowing that there is only INTR helps. My understanding was that it was just wires not the true interrupt register. So I will need to clear INTR if polling instead of ISR which uses INTSx.

lurch, 2022-11-25:
This will be fixed in the next release of the PDF datasheets.

closed 2022-11-25 by lurch