raspberrypi/pico-feedback #286: Datasheet: Order Codes

attie-argentum, 2022-11-01:
The datasheet (v1.9) currently lists a handful of order codes for the Pi Pico:

[image]

- "_Raspberry Pi Pico H_" is easy enough to deduce as "_with Headers_" ([ref](https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html)), though there isn't any reference to it in the datasheet at all - no text, photos, dimensioned drawings, etc...
- `SC0915` and `SC0916` apparently differentiate the country of origin ([thanks Alasdair!](https://twitter.com/aallan/status/1587545515515957248)), but there is no mention of this in the datasheet.
    - As I understand it they are 100% identical apart from CO (which-is-which is still a mystery to me)
    - `5056561801445` issuing country: UK
    - `0617588405587` issuing country: US (not Japan?)

Please could we have some clarity on this front?

aallan, 2022-11-05:
> * "_Raspberry Pi Pico H_" is easy enough to deduce as "_with Headers_" ([ref](https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html)),  though there isn't any reference to it in the datasheet at all - no text, photos, dimensioned drawings, etc...

It's on the list of things to do, but fairly low down the list, as the difference between the boards isn't huge except for the addition of headers to the Pico H. Which is fairly self explanatory.

> * `SC0915` and `SC0916` apparently differentiate the country of origin ([thanks Alasdair!](https://twitter.com/aallan/status/1587545515515957248)), but there is no mention of this in the [...]

As I said on Twitter, one is produced in Japan, the other in the UK. The boards are identical in all respects apart from the EAN written on the label on the reels.

> Please could we have some clarity on this front?

Not sure what additional clarity is needed? Folks that care about the order code will almost certainly be talking directly to our commercial team. They're honestly there just so I don't get confused when one specific internal spreadsheet comes across my desk yet again.

closed 2022-11-05 by aallan

attie-argentum, 2022-11-06:
> The boards are identical in all respects apart from the EAN

Understood

> Not sure what additional clarity is needed?

I was hoping for clarity in the datasheet... just an extra column or two, example below.
It's a simple fix and removes any confusion on this front.

| Model | Order Code | EAN | MOQ | RRP | Country of Origin | Notes |
| --- | --- | --- | --- | --- | --- | --- |
| Raspberry Pi Pico |  SC0915 | 5056... | 1+ | US$4.00 | UK | -
| Raspberry Pi Pico |  SC0916 | 0617... | 1+ | US$4.00 | Japan | -
| Raspberry Pi Pico H |  SC0917 | 5056... | 1+ | US$5.00 | UK | With headers variant

> Folks that care about the order code will almost certainly be talking directly to our commercial team

Maybe... but I tripped on this and couldn't figure it out, who's to say it's not unnecessarily consumed time for others too?

> They're honestly there just so I don't get confused when one specific internal spreadsheet comes across my desk yet again.

If they're customer-facing MPNs, then it's not just for you.

JamesH65, 2022-11-06:
Given that order codes can change over time anyway, or could be extended (manufacture in another country, batch quantities) I'd suggest removing the codes (order, EAN) completely just so this document doesn't need updating in the future. Not really necessary for end users to need those, more a reseller/commercial thing.

attie-argentum, 2022-11-06:
> Not really necessary for end users to need those, more a reseller/commercial thing.

I disagree.

If we're taking about "end users" being hobbyists or low volume users who want "a Raspberry Pi Pico", then sure, it's not particularly useful.

But if we're taking about more formal designs that specify the MPN on the BOM, then this info is important - especially when that BOM is passed to colleagues to conduct the purchasing. Being able to list alternate parts is also important (in this case, both are acceptable). Clarity in communication, etc...

aallan, 2022-11-06:
I'm not intending to make changes to the datasheet.

attie-argentum, 2022-11-06:
> I'm not intending to make changes to the datasheet.

Unexpected... but okay, thanks.