raspberrypi/pico-feedback #288: I2C core bus clock "ic_clk" not defined in RP2040 Datasheet.

VBruceHunt, 2022-11-02:
In the **RP2040 Datasheet** with build-date: _2022-06-17_ and build-version: _7028f9f-clean_ the table of contents, "4.3.14. **IC_CLK** Frequency Configuration" is described but its source is never defined.  No mention of it appears in any of 2.15 Clocks, 2.16 Crystal Oscillator (XOSC), 2.17 Ring Oscillator (ROSC) or 2.18 PLL.  The source and possible values of this clock is critically important to proper setup of the I2C peripherals especially for high speed plus I2C bus operation.  Several registers and equations in the I2C peripheral documentation depend upon this value: i.e. _4.3.14.1. Minimum High and Low Counts in SS, FS, and FM+ Modes_, _4.3.14.2. Minimum IC_CLK Frequency_, and  _4.3.14.3. Calculating High and Low Counts_.  

Observed Typos in the I2C Peripheral Section ( 4.3 )

- 4.3.7.1 :  Figure 72,  the label "CDM" should be "CMD".  The meanings of the "Stop" and "Restart" bits are ambiguous where the bit values should be explicit. 
- 4.3.10.1:  The first appearance in the text of "ic_clk" appears with no explanation. 
- 4.3.14:  This is factually incorrect.  The bus may be shorted or a bus interface component unrelated to this RP2040 I2C module may have failed.  No amount of resetting this I2C peripheral will clear the fault.  It is best to suggest a reset of this I2C module to see if it will clear the bus; but also to state that further physical intervention may be required.
- 4.3.16 Operation of Interrupts Registers is NOT about Registers.  There is no description of a register; but a table of interrupt signals and whether they are cleared by software or hardware.  The state of the I2C module and the I2C bus that causes a specific interrupt should be described along with how each interrupt signal is cleared.

lurch, 2022-11-02:
As hinted at by the start of section 4.3, the majority of the I2C documentation has been copied from the Synopsys documentation for the I2C block.

> 4.3.7.1 : Figure 72, the label "CDM" should be "CMD".

Already reported in #250 

> 4.3.10.1: The first appearance in the text of "ic_clk" appears with no explanation.

Section 4.3.1.2 says "The I2C controller is connected to clk_sys. The I2C clock is generated by dividing down this clock, controlled by registers inside the block."

> 4.3.16 Operation of Interrupts Registers is NOT about Registers.

I suspect this probably needs to be read in conjunction with 4.3.17 where it _does_ talk about the interrupt registers?

lurch, 2022-11-02:
> The I2C controller is connected to `clk_sys`.

In the next release of the PDF, this will say "All clocks in the I2C controller are connected to `clk_sys`, including `ic_clk` which is mentioned in later sections."