raspberrypi/pico-feedback #290: Adding VSCode Pico-W-Go extension as alternative to Thonny to the Pico docs chapter MicroPython

paulober, 2022-11-09:
Is it possible to list my Visual Studio Code extension (Pico-W-Go) as an alternative to Thonny in the Raspberry Pi Pico (W) documentation in the MicroPython chapter?

Pico-W-Go is a Visual Studio code extension designed to simplify and speed up the development of MicroPython projects for the Raspberry Pi Pico and Pico W boards.
More information can be found here: https://github.com/paulober/Pico-W-Go or in the wiki https://github.com/paulober/Pico-W-Go/wiki.

I thought that, since your documentation is very much designed to help beginners getting started with MicroPython microcontroller programming, it would be helpful for some of them to be able to use the editor they probably already know for programming.

The Visual Studio Code extension already has over 5.1 thousand downloads. I have already received constructive feedback, which helped me to improve the extension. I plan to continue to build on this in the future. The link to the Microsoft VS Code marketplace publication is as follows: https://marketplace.visualstudio.com/items?itemName=paulober.pico-w-go

(Just as a quick note concerning the contributing guidelines third party chapter: I was referred to this repository by a Raspberry Pi Foundation staff member with this request.)

aallan, 2022-11-09:
Something like 85% of Raspberry Pi boards in the wild run 32-bit user space Raspberry Pi OS.

paulober, 2022-11-09:
@aallan Happy to see that this has already been discussed internally and is planned.

@lurch I would like to. The reason why the extension is limited to specific architectures is the node serialport library and as you can see on their website and in my extension this dependency includes prebuilds also for linux armv6 and armv7 the 32-bit arm versions which will be used by a raspberry pi running in "32-bit mode" on the armv8 processors. I'm a bit confused that these are not working as some community members told me. 
I'm setting up a Pi4b in 32-bit and see what i can find out. I'll send an update on this once i figured it out.

paulober, 2022-11-10:
@aallan @lurch Just published version [v2.1.3](https://github.com/paulober/Pico-W-Go/releases/tag/v2.1.3) which added support for armv7 (aka Raspberry Pi OS 32-bit on 3b+, 4b or 400).

lurch, 2022-11-10:
> armv7 (aka Raspberry Pi OS 32-bit on 3b+, 4b or 400).

Also the Pi 2B and the Pi Zero 2 W :slightly_smiling_face:  https://www.raspberrypi.com/documentation/computers/processors.html

paulober, 2022-11-10:
> > armv7 (aka Raspberry Pi OS 32-bit on 3b+, 4b or 400).
> 
> Also the Pi 2B and the Pi Zero 2 W 🙂 https://www.raspberrypi.com/documentation/computers/processors.html

True, should theoretically work just as well as on the other platforms, but unfortunately can't test this myself on these.

paulober, 2023-01-05:
@aallan Any updates on this?

aallan, 2023-01-05:
> @aallan Any updates on this?

Not really. The toolchain problems are now sorted, it's just time to sit down and write some documentation around it that's needed. It's on the list of things to do.

ghollingworth, 2023-02-03:
Paul,

Will you please get in touch with me at gordon@raspberrypi.com

Gordon

aallan, 2023-03-02:
Some information about the extension and using it with VSCode will be added in the next documentation release, which should happen this week.

closed 2023-03-02 by aallan

paulober, 2023-03-02:
> Some information about the extension and using it with VSCode will be added in the next documentation release, which should happen this week.

Happy to hear it. Will check the documentation every now and then.

paulober, 2023-06-03:
@aallan 
Apologies for reaching out once more. I seem to be unable to locate the note regarding Pico-W-Go that you mentioned in the Raspberry Pi Docs on the website. Could you kindly assist me in finding it?
Thank you.

aallan, 2023-06-03:
It's not on the website, it's in the PDF documentation, see Section 4.2 of https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-python-sdk.pdf.

paulober, 2023-08-01:
Hey @aallan,

Just wanted to give you a heads up on some changes to the extension:

1. The FTP-Server is removed - it was outdated. Instead, we now have a new Virtual FileSystem/Workspace integration into VSCode.
2. I renamed the extension to "MicroPico" to better match its focus on supporting MicroPython projects for Raspberry Pi Pico and Pico W boards. And it's ID to `paulober.micropico`