raspberrypi/pico-feedback #292: [Feature Request] provide RP2040 data sheet as HTML

JustAnother1, 2022-11-19:
It would be nice to have the datasheet as a collection of HTML pages. This would allow users to give out links to chapters inside the data sheet. This would help in discussions about features. 

Finding the needed information might also be faster as people are used to find information on web pages and would not need to download the big PDF file before they can even start looking.

pidd-wbiz, 2022-11-20:
> It would be nice to have the datasheet as a collection of HTML pages. This would allow users to give out links to chapters inside the data sheet. This would help in discussions about features. [...]

Its detailed how to create html documentation on the page [https://github.com/raspberrypi/documentation](url)

JustAnother1, 2022-11-20:
@pidd-wbiz T have tested that, but the generated HTML only links to the PDF of the RP2040 data sheet.

What I want is the pages of the datasheet as HTML so that I can link to the UART, SPI or PIO description directly. Or set a link to the register that can be read to detect on which core the firmware is running.

Right now I can only link to the RP2040.pdf file and explain in which section or on which page to look.

So thanks for the suggestion, but that does not solve my issue.

aallan, 2022-11-20:
While the RP2040 and Raspberry Pi Pico PDF documentation is generated from AsciiDoc source, in a similar fashion as the [online documentation](https://www.raspberrypi.com/documentation/), they do not share a toolchain. There would be quite a lot of work involved in creating an online version of the PDF documentation that makes sense. 

While it's on my "future roadmap" as a possible feature, it's not going to be something that is going to happen in the near term. Sorry.

closed 2022-11-20 by aallan