raspberrypi/pico-feedback #295: Diagram pinout in pico-examples/i2c/ssd1306_i2c/ is wrong

thisisgregm, 2022-11-25:
Seems like pinout on the diagram picture is shifted by one pin.

SDA, SCL should be pin 6 and 7 respectively
VCC, GND should be 36 and 38 respectively.

But the diagram shows the connections to pins 7, 8, 35 and 37 respectively.

I've noticed that as pin 8 is GND, but it's connected to SCL on the diagram :)

Luckily the description of pins in ssd1306_i2c.c is correct.

If you shift the Pico by one hole left all will be fine I think.

peterharperuk, 2022-11-25:
Looks like this should be a comment on this pull request? https://github.com/raspberrypi/pico-examples/pull/217
@JamesH65

JamesH65, 2022-11-25:
Have asked for that PR to be merged, but will try and fix the issue before that happens.

JamesH65, 2022-11-25:
So, I have fixed the PR. Entire Pico had been shifted right in Fritzing.

JamesH65, 2022-11-25:
And the linked PR has now been merged, so this is fixed.

closed 2022-11-25 by JamesH65