raspberrypi/pico-feedback #301: Incorrect timing diagrams in chapter 3.2.2

mtexplorative, 2023-01-01:


lurch, 2023-01-01:
I assume it's this one https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf

See also the big discussion in #121

mtexplorative, 2023-01-01:
Sorry, I refer to the pico--c-sdk:
https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf

mtexplorative, 2023-01-01:
@lurch thank you for the hint. Seems that the issue 'wrong figure' is already around some time. I just struggled when I was reading the German c't Make: "PI Pico Special". I struggled understanding the figures they included in their article about PIO. When asking the authors for helping me in better understanding, they were pointing to the original documentation, the mentioned raspberry-pi-pico-c-sdk.pdf.

At least for me, the correction of the figures would help me in better understanding the effects of `side x`.

aallan, 2023-09-05:
@nathan-contino Check with the ASIC team (and probably @jackbenwillis) and fix up or close appropriately.