raspberrypi/pico-feedback #305: why do i get this error " File "<stdin>", line 12, in <module>"?

ashas1005, 2023-01-21:


lurch, 2023-01-21:
Without more details, it's impossible to provide any help...
What were you trying to do when you got this error? Which program is giving you this error? What code are you using? etc. etc.

ashas1005, 2023-01-21:
Sir,
Thanks for the mail. Requested data is as follows:
i am a beginner in this area trying to learn Raspberry Pi Pico. i
was interfacing an OLED display SSD1306 based using I2C. . The connections
are as follows GP8 (Pin 11) of Pico connected to SDA (Pin 4) and GP9(Pin
12). I installed the library files microphython-ssd1306 and
microphyton-oled from the "manage packages" item on the THORNY screen as
described. i have enclosed both the files for your reference, the following
screen shot will give further details on the error message.
>>> %Run -c $EDITOR_CONTENT
Traceback (most recent call last):
  File "<stdin>", line 12, in <module>
  File "/lib/oled/ssd1306.py", line 127, in __init__
  File "/lib/oled/ssd1306.py", line 47, in __init__
  File "/lib/oled/ssd1306.py", line 72, in init_display
  File "/lib/oled/ssd1306.py", line 132, in write_cmd
OSError: [Errno 5] EIO
>>>

i have tried several times but have been getting the same error message.
Please help me understand how to solve the issue and help me in
understanding the mistake here.

Thanks and Best Regards

SUBRAHMANYAM SISTA

lurch, 2023-01-22:
For questions about MicroPython you should ask for help at https://github.com/orgs/micropython/discussions/categories/ports-rp2040-pico

closed 2023-01-22 by lurch