raspberrypi/pico-feedback #306: Debugging Raspberry Pi Pico with pico probe "getting-started-with-pico.pdf" issues

Mitch6886, 2023-01-23:
I am using windows to debug the pico probe, I obtained the UT2 file from "https://datasheets.raspberrypi.com/soft/picoprobe.uf2". I did use zadig to change the driver several times

In section Appendix A>Build OpenOCD>Windows, I needed to use "Mingw64.exe" which was not clearly described I thought I shouuld use "Msys.exe"

In section Appendix A>Using Pico Probe with OpenOCD, I needed to use "interface/picoprobe.cfg" instead of "interface/cmsis-dap.cfg", is the UT2 file hosted on the raspberry pi site out of date? There did not seem to be instructions for how to build it on windows.

tobybroberts, 2023-02-28:
In light of the recent roll out of the Raspberry Pi Pico Windows Installer can this issue be closed?

aallan, 2023-03-02:
This will be resolved in the latest PDF documentation release due this week.

closed 2023-03-02 by aallan